warhammerkid/bluetti_mqtt #2: unattended way to run the program

dabene1408, 2022-06-02:
Hi, not a bug but a question
I have no idea how i run your program unattended on my Raspberry.
How would you do it?
Greets

warhammerkid, 2022-06-04:
I don't currently have a raspbian system to verify this with, but once I do I'll update the README with instructions.

If you ran `pip install bluetti_mqtt` then you should have a `bluetti-mqtt` command installed (you might have needed to use `pip3`). On Ubuntu it's in `/home/$USER/.local/bin/`, but you can find where it is by running `which bluetti-mqtt`. If that doesn't work, run `python -m site --user-base` and look in the `bin` subfolder.

```
$ python -m site --user-base
/home/ubuntu/.local
$ ls /home/ubuntu/.local/bin/
bleak-lescan    bluetti-logger    bluetti-mqtt
```

Once you have the path to the `bluetti-mqtt` command you can use that in a systemd service. This file should be placed at `/etc/systemd/system/bluetti-mqtt.service`. You'll need to update the path, MQTT broker host, and bluetooth addresses.

```systemd
[Unit]
Description=Bluetti MQTT
After=network.target
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=30
TimeoutStopSec=15
User=pi
ExecStart=/PATH/TO/bluetti-mqtt --broker BROKER_HOST_HERE BLUETOOTH_ADDRESS
KillSignal=SIGINT

[Install]
WantedBy=multi-user.target
```

Once you've created the file, you should be able to run `sudo systemctl enable bluetti-mqtt`, so that it automatically starts when you reboot, and `sudo systemctl start bluetti-mqtt` to start it. If that doesn't work, try running `sudo journalctl -u bluetti-mqtt` to see if it printed out any errors.

dabene1408, 2022-06-06:
Thanks, now it's working :)

closed 2022-06-06 by dabene1408