warhammerkid/bluetti_mqtt #4: More values

chgrkl, 2022-07-15:
Hello,
First of all thanks for the great project!
I've had an AC200MAX for about 6 weeks.
Since then it has been running on a RaspberryPi and sends to FHEM via mqtt :-)
Yesterday my B230 extension battery arrived and of course I connected it right away =)
With the AC200MAX and in the app I can see the status of the batteries individually (installed and extended) in percentage.
Unfortunately, no further information arrives in the MQTT device.
Here comes only the percent and total_battery_percent, both equal to total percent value of AC200MAX.
Can you extend bluetti_mqtt to that effect?

Regards

Christoph

warhammerkid, 2022-07-17:
Since I personally only have a single battery pack for my AC300 to test, the first step in addressing this would be for you to install the just-released v0.6.1 and run the `bluetti-logger` command. Ideally you would run this when the two packs are at different states of charge, or if that’s not possible due to how charging/discharging work, while you are actively charging or discharging the battery. Ideally you’ll run the logger long enough for both packs to change their state of charge by at least 1%. Once you have the log I would recommend emailing it to me (you can find my email address on my Github profile), since I’m not 100% certain the device serial number should be shared publicly.

warhammerkid, 2022-07-25:
@chgrkl Looking over the log file you sent me revealed two things:
- There seems to be a longer delay than I expected after telling the device to give me information about the other battery packs before it's actually available.
- I'm apparently so bad at Python that I made a novice error and managed to only get the first _two_ batteries, but the information I needed was from the third battery. :man_facepalming:

Version 0.6.2 has been released to address both these issues, so I'll need you to re-run the logging command after updating so that I have the information from both battery packs.

Based on the data you sent and past data I've collected, it looks like I should be able to get at least the individual battery percentages and cell voltages. Beyond that, I'll just have to see - there should be a BMS version, full pack voltage, and possibly capacity information somewhere in the data...

PR #17 (2022-08-29): Multiple pack support
Adds API for configurable device-specific polling behavior (optimized separately from logging), as well as parallel pack and non-pack data polling loops.

Fixes #4
closed 2022-08-29 by warhammerkid