warhammerkid/bluetti_mqtt #5: Support EB3A

warhammerkid, 2022-07-17:
In theory basic support should be as simple as updating my device name regex to recognize this device. I don’t own one currently, though, so I have no way to test for sure. Also, without bluetooth captures of the app, I don’t have any way to discover how device-specific functionality (for example the light) works.

jretz, 2022-08-09:
I have an EB3A. Adding `EB3A` to the `DEVICE_NAME_RE` regex does cause it to show up in scans (i.e., it's the right prefix) and it does get some data from the device.

It only seems to get one round of (partial?) data published to MQTT. After I ctrl-c out of `bluetti-mqtt.py`, I see that there was an unhandled ValueError in an async task because there's no `AutoSleepMode` enum value for `0`. So I tried this:

```
@unique
class AutoSleepMode(Enum):
    WHO_KNOWS = 0  # Placeholder to get past the ValueError
    THIRTY_SECONDS = 2
    ONE_MINUTE = 3
    FIVE_MINUTES = 4
    NEVER = 5
```

With that, it continues to send updates to MQTT and there are no exceptions on ctrl-c. The enum values for `AutoSleepMode` look like the screen timeout options on my AC300. I'm not aware of any way to set the screen timeout on the EB3A itself or in the app - it seems hard coded at 30 seconds.

Without an `--interval` specified, my AC300 updates nearly continuously (more than once a second). My EB3A seems to update every 30 seconds.

Let me know if I can help further. I also have a split phase 2 x (AC300 + B300) setup that is my real use for this very cool project, but the EB3A is a fun place to start. I see the logger script... would it be helpful if I run that while turning subsystems on/off on the EB3A? I know my way around Python (including some rusty async knowledge), but not Bluetooth (other than as a user).

I won't be able to do much with the EB3A for a few days... I just tried to update firmware and it failed, leaving my EB3A non-functional. Looking online, people seem to have some success with letting the battery die, then charging it and trying again. My battery was at 90%, so it will probably take most of the week. Hopefully it'll be back soon. 🤞

jschoch, 2022-08-10:
I can confirm that these fixes work on the eb3a

[image]

when I run the log i get

```
(base) PS C:\Users\jesse\Documents\PlatformIO\Projects\bluetti_mqtt> bluetti-logger --log log.txt 04:7F:0E:A1:68:05
Connecting to 04:7F:0E:A1:68:05
Waiting for connection...
Device has a maximum of 1 battery packs
Needed to disconnect due to error: too many retries
Needed to disconnect due to error: too many retries
```

and the log looks like this

```
tail log.txt 
{"type": "client", "time": "2022-08-10 07:00:57 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-10 07:23:13 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-10 07:23:48 -0700", "data": "AQOEAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAGrgAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAZeg=", "command": "AQMARgBCJC4="}
{"type": "client", "time": "2022-08-10 07:23:49 -0700", "data": "AQOUAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKWb", "command": "AQMAiABKRBc="}```

jschoch, 2022-08-10:
i plugged in the AC (the solar was plugged in before) and it appears much is not updating.

[image]

Happy to provide a capture if you can help me get the logging working.

warhammerkid, 2022-08-15:
@jschoch @jretz Can you check and see if the new branch I put up fixes logging? The snippet of logs that Jesse posted don't include the lower status page, so I made some tweaks to how I'm querying it that might fix things (assuming that it is indeed not working right now). I also updated the `bluetti-logger` command so that it'll output the parsed data to the console in addition to writing the raw response data to the log file.

jschoch, 2022-08-16:
The branch will connect out of the box but I still get the logging error

```
(base) PS C:\Users\jesse\Documents\PlatformIO\Projects\bluetti_mqtt> bluetti-logger --log log.txt 04:7F:0E:A1:68:05
Connecting to 04:7F:0E:A1:68:05
Waiting for connection...
Device has a maximum of 1 battery packs
Needed to disconnect due to error: too many retries

```

the log 
```
tail log.txt 
{"type": "client", "time": "2022-08-10 07:00:57 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-10 07:23:13 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-10 07:23:48 -0700", "data": "AQOEAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAGrgAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAZeg=", "command": "AQMARgBCJC4="}
{"type": "client", "time": "2022-08-10 07:23:49 -0700", "data": "AQOUAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKWb", "command": "AQMAiABKRBc="}
{"type": "client", "time": "2022-08-10 07:46:53 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}

```

warhammerkid, 2022-08-18:
Bah, what a pain this is to debug. I changed it again and pushed up a change. If you want to poke at this yourself, the issue is this bit of code in the `logger_cli.py` file:

```python
await log_command(client, device, QueryRangeCommand(0x00, 0x0A, 0x35), log_file)
```

I have no idea what values for the second and third argument (the start of the range and then length of the range) will actually work, although it looks like somebody in the diysolarforum thread did post that `0x0A, 0x35` worked for them. If you're still getting the disconnect error I'd advise lowering the `0x35` bit by bit until it starts working.

jschoch, 2022-08-18:
Some progress, i had to turn off the pack polling

```
(base) PS C:\Users\jesse\Documents\PlatformIO\Projects\bluetti_mqtt> bluetti-logger --log log.txt 04:7F:0E:A1:68:05
Connecting to 04:7F:0E:A1:68:05
Waiting for connection...
logcmd
Result from log_command
{'device_type': 'EB3A', 'serial_number': 2224002537549, 'dc_input_power': 0, 'ac_input_power': 0, 'ac_output_power': 0, 'dc_output_power': 0, 'power_generation': Decimal('0.5'), 'total_battery_percent': 72, 'ac_output_on': False, 'dc_output_on': True}
0x3d
Result from log_command
{'ac_output_on': False, 'dc_output_on': True}
Got an error running command UpdateFieldCommand(page=0x0b, offset=0xbe, value=0x01): too many retries
Result from log_command
{'ac_input_voltage': Decimal('0'), 'dc_input_voltage': Decimal('167.5')}
Result from log_command
{}
logcmd
Result from log_command
{'device_type': 'EB3A', 'serial_number': 2224002537549, 'dc_input_power': 0, 'ac_input_power': 0, 'ac_output_power': 0, 'dc_output_power': 0, 'power_generation': Decimal('0.5'), 'total_battery_percent': 72, 'ac_output_on': False, 'dc_output_on': True}
0x3d
Result from log_command
{'ac_output_on': False, 'dc_output_on': True}
```

tail log.txt 
{"type": "client", "time": "2022-08-18 07:10:05 -0700", "data": "AQOEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAF2gAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPDI=", "command": "AQMARgBCJC4="}
{"type": "client", "time": "2022-08-18 07:10:05 -0700", "data": "AQOUAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKWb", "command": "AQMAiABKRBc="}
{"type": "client", "time": "2022-08-18 07:10:59 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-18 07:11:34 -0700", "data": "AQOEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAF8AAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAnf4=", "command": "AQMARgBCJC4="}
{"type": "client", "time": "2022-08-18 07:11:34 -0700", "data": "AQOUAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKWb", "command": "AQMAiABKRBc="}
{"type": "client", "time": "2022-08-18 07:12:58 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-18 07:14:17 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-18 07:16:49 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}
{"type": "client", "time": "2022-08-18 07:21:05 -0700", "data": "AQNqRUIzQQAAAAAAAAAAA/uYTdDhAgUAAAAAAAAjjwADIyoAAwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABQAAAEgAAQAAAAAAAAAAAAEAAAABAAAAAgCAAAAAAAAAAAAAAAAAAAAAAIlN", "command": "AQMACgA1pd8="}
{"type": "client", "time": "2022-08-18 07:21:05 -0700", "data": "AQN6AAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADaUw==", "command": "AQMLuQA9V9o="}

```

PR #11 (2022-08-20): Device specific parsing
The current implementation assumes that all devices have the same fields and the same parsing for those fields. However, not every device supports all functionality, and not every device encodes field data the same way (AC200M has different decimal scales for several fields). This PR addresses this by adding a declarative, reusable parser API for building device-specific parsers.

Initially, the downside of this is additional boilerplate and copied code. However, over the long term there should be minimal need to change existing devices once the initial flurry of discov [... cut]
closed 2022-08-20 by warhammerkid