warhammerkid/bluetti_mqtt #15: Scan does not working

mwende, 2022-08-22:
Hi i have installed with pip command but weh i will mache a scan for my devices i get the following error.

Traceback (most recent call last):
  File "/home/mwende/.local/bin/bluetti-mqtt", line 8, in <module>
    sys.exit(main())
  File "/home/mwende/.local/lib/python3.8/site-packages/bluetti_mqtt/server_cli.py", line 123, in main
    cli.execute()
  File "/home/mwende/.local/lib/python3.8/site-packages/bluetti_mqtt/server_cli.py", line 62, in execute
    asyncio.run(scan_devices())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/mwende/.local/lib/python3.8/site-packages/bluetti_mqtt/bluetooth/__init__.py", line 80, in scan_devices
    devices = await BleakScanner.discover()
  File "/usr/local/lib/python3.8/dist-packages/bleak/backends/scanner.py", line 116, in discover
    async with cls(**kwargs) as scanner:
  File "/usr/local/lib/python3.8/dist-packages/bleak/backends/scanner.py", line 96, in __aenter__
    await self.start()
  File "/usr/local/lib/python3.8/dist-packages/bleak/backends/bluezdbus/scanner.py", line 128, in start
    manager = await get_global_bluez_manager()
  File "/usr/local/lib/python3.8/dist-packages/bleak/backends/bluezdbus/manager.py", line 812, in get_global_bluez_manager
    await instance.async_init()
  File "/usr/local/lib/python3.8/dist-packages/bleak/backends/bluezdbus/manager.py", line 294, in async_init
    bus = MessageBus(bus_type=BusType.SYSTEM)
  File "/usr/local/lib/python3.8/dist-packages/dbus_next/aio/message_bus.py", line 122, in __init__
    super().__init__(bus_address, bus_type, ProxyObject)
  File "/usr/local/lib/python3.8/dist-packages/dbus_next/message_bus.py", line 85, in __init__
    self._setup_socket()
  File "/usr/local/lib/python3.8/dist-packages/dbus_next/message_bus.py", line 575, in _setup_socket
    raise err
  File "/usr/local/lib/python3.8/dist-packages/dbus_next/message_bus.py", line 548, in _setup_socket
    self._sock.connect(filename)
FileNotFoundError: [Errno 2] No such file or directory

Can you tell me whats is wrong...

warhammerkid, 2022-08-24:
This sounds like either BlueZ or DBUS is not installed (or somehow not installed correctly?). What Linux distro are you using?

mwende, 2022-08-24:
i Use Ubuntu 22.04.1 LTS on an Proxmox VM... DBUS is installed and running.
I try your Script on diffrent Clients WSL, Proxmox and diffrent distibutions

warhammerkid, 2022-08-29:
It appears that by default Proxmox will not pass access to the Bluetooth hardware to VMs. Have you configured it to pass client access to whatever Bluetooth device you are using? If you run `hcitool dev` inside your Ubuntu VM, does it list a device?

```
$ hcitool dev
Devices:
        hci0    00:11:22:33:44:55
```

It appears that WSL has no support for Bluetooth at this point in time, so that won't work. However, if your host machine is Windows you should be able to install Python and run it there if you'd rather not deal with the complication of a VM.

mwende, 2022-08-29:
Hi, thanks... now i habe set the parameter for my AC300 an it works fine....

closed 2022-08-29 by mwende