warhammerkid/bluetti_mqtt #20: Data just from one power station

dabene1408, 2022-08-29:
When I try to use your script with multiple power stations by providing (in my case 2) MAC-Addresses, only one is read. If I restart the service that's responsible for your script multible times, sometimes i get data from the other power station. The data of both power stations is read maby every 10 restarts for one cycle.

dabene1408, 2022-08-31:
I use version 0.8.2 on a Raspberry Pi 4 with Raspbian and try to read two EP500P power stations.

warhammerkid, 2022-10-23:
I finally have a second device to test with (an EB3A). I ran the latest version for over 10 hours without any issues monitoring both an AC300 and an EB3A. My monitoring device is a Raspberry Pi 4, also running Raspbian, and I wasn't using the `--interval` flag to limit how quickly it polls.

There's a bug fixed in 0.9.0 that might have fixed things, as well as newer versions of [bleak](https://github.com/hbldh/bleak), the library I'm using for bluetooth. It might be worth running `pip install -U bleak` if you haven't in a while, to upgrade.

dabene1408, 2022-10-23:
It seems the update helped. I'll monitor and test it through the next days. 
Thanks for the help.