warhammerkid/bluetti_mqtt #22: Bluetti_mqtt limited to homeassistant or compatible with other mqtt servers/home automation platforms?

xdaamg, 2022-08-31:
No issue, just a question. I'm thinking about buying a bluetti ac200max. But an api would be essential, cause i would use it also in my home in combination with "iobroker" home automatisation. I would for example automate, when charging should start.

"Bluetti_mqtt" is not limited to homeassistant, right? It should work also with other mqtt servers, like iobroker mqtt, correct?

lbossle, 2022-08-31:
This library should work with any MQTT broker. The compatibility with Home Assistant only means that the device will automatically appear in HA. This will not be the case for other home automation systems. You'd need to set that up yourself.

Regarding your automation idea, I don't thinks it's possible to toggle AC/DC inputs on the Bluetti AC200 Max itself?

bofi99, 2022-08-31:
Hello,

MQTT works since release 0.6 (mqtt user +password) with iobroker. I automatically switch  AC via mqtt command depending on SOC.

warhammerkid, 2022-08-31:
> I would for example automate, when charging should start

There is no way that I know of to do this with the AC200M. There is no API for controlling whether it should charge or not if plugged into a power input.

> It should work also with other mqtt servers, like iobroker mqtt, correct

Correct. It’s just a normal MQTT client.

xdaamg, 2022-09-01:
Thank you for this information.
I would use a shelly 1 pm device to power on/off ac charger, depending on SOC and power from my solar system.

xdaamg, 2022-09-12:
States in iobroker mqtt are created as expected, but command objects are not created. Is this a bug or it should be like this?
I used iobroker/javascript code below to create an object for AC and DC Switch under "bluetti/command". It's possible to switch ac and dc output now (GUI), maybe not best solution, but working. @bofi99 Could you post your code?

`on({id: 'javascript.0.Strom.Bluetti.AC200M-xxx.dc_output_on', change: "ne"},
    function(test) {

//Init Command Object, comment out, after init
/*
    var obj = {
        "type":"state",
        "common": {
        "name": "bluetti/command/AC200M-xxx/dc_output_on",
        "role": "variable",
        "type": "string",
        "read": true,
        "write": true,
        "desc": "Bluetti DC Output",
        "def": "uninitialized"}
    };

    setObject('mqtt.0.bluetti.command.AC200M-xxx.dc_output_on', obj, function() {
        setState('mqtt.0.bluetti.command.AC200M-xxx.dc_output_on', obj.common.def, true);
    });
*/

        var status_dc_output = getState('mqtt.0.bluetti.state.AC200M-xxx.dc_output_on').val;
        var status_dc_output_soll = getState('javascript.0.Strom.Bluetti.AC200M-xxx.dc_output_on').val;

        if (status_dc_output_soll == true) {
            if (status_dc_output == "OFF") {
                //sendTo("mqtt.0", "sendMessage2Client", { "topic": 'bluetti/state/AC200M-xxx/dc_output_on', "message": 'ON'});
                setState('mqtt.0.bluetti.command.AC200M-xxx.dc_output_on', 'ON');
            }
        } else {
            if (status_dc_output == "ON") {
                //sendTo("mqtt.0", "sendMessage2Client", { "topic": 'bluetti/state/AC200M-xxx/dc_output_on', "message": 'OFF'});
                setState('mqtt.0.bluetti.command.AC200M-xxx.dc_output_on', 'OFF');
            }
        }
});`

bofi99, 2022-09-16:
Hello,

i still have version 0.6 installed because it works perfect. I use an blockly-script in iobroker to set the mqtt value to on or off:

var Inverter;

Inverter = getState("mqtt.0.bluetti.command.AC200M-XXXX.ac_output_on").val;
on({id: [].concat(['mqtt.0.bluetti.state.AC200M-XXXX.total_battery_percent']), change: "ne"}, async function (obj) {
  var value = obj.state.val;
  var oldValue = obj.oldState.val;
  if ((obj.state ? obj.state.val : "") <= 10) {
    Inverter = 'OFF';
    sendTo("telegram.0", "send", {
        text: 'Inverter aus'
    });
    console.log('Akku aus!');
  }
  if ((obj.state ? obj.state.val : "") >= 80) {
    Inverter = 'ON';
    sendTo("telegram.0", "send", {
        text: 'Inverter an'
    });
    console.log('Akku an!');
  }
  if (Inverter != getState("mqtt.0.bluetti.command.AC200M-XXXX.ac_output_on").val) {
    setState("mqtt.0.bluetti.command.AC200M-XXXX.ac_output_on"/*bluetti/command/AC200M-XXXXX/ac_output_on*/, Inverter);
  }
});

closed 2022-09-19 by xdaamg