warhammerkid/bluetti_mqtt #23: Bundle as an HA Integration?

joeladria, 2022-09-02:
Hello, thanks for the wonderful work on this. 

Just wondering if there is a bundling of this script for running as an Integration on HA itself, via HACS or similar.

I was thinking of doing up a little container like was done for rtl_433 https://github.com/pbkhrv/rtl_433-hass-addons

mikedgit, 2022-09-02:
Joel, I think the work that @warhammerkid has been doing on functionally breaking up code is really going to help enable this.  While a container would be a quick solution, I'd like to see an integration like you mention more - ESPECIALLY now that bluetooth is top notch with the latest HA releases.

The biggest issue with WiFi based integration is that you have to intercept the comms between Bluetti and its home base.  It's much cleaner to have a bluetooth integration (albeit with much lower range between your HA instance and the Bluetti).

`<tangent>`
The fact that this is leading to a ESP32 integration is pretty interesting to me because you get the best of both worlds (i.e. bluetooth comms with Bluetti, and wifi comms to HA or other).  Also could potentially use up and coming Thread networking with Espressif's ESP32-H2 with Thread support.
`</tangent>`

Tangent aside, a bluetooth-based HA integration (no ESP required) would be AWESOME.  I wouldn't be surprised if it was allowed in the official integration list (not just HACS).  HA Add-ins work but are more heavy-handed....

tommy99999999, 2022-09-07:
Joel, you have my vote. I would like to see some sort of HA integration.
Awesome!

twister36, 2022-10-01:
I would love this too.. the only way I've gotten this to work is by running this separately and outside of HA. Thanks for all the hard work!

warhammerkid, 2022-10-09:
Started poking at writing a custom bluetooth integration this week, and it looks like a TON of work. As far as I can tell, documentation is weak, and a lot of the official integrations just seem to be copying code from each other, so it's hard to tell _why_ things are the way they are. A more experienced HA integration developer would hopefully be able to work through this faster than I can, but I'll keep poking along...

First step was to build a config flow and hook up bluetooth discovery, which after a few issues, produced some success:

[image]

It's nice to see that HA is using the `bleak` library for bluetooth, so the tweaks I'll need to make to reuse the work I've done so far shouldn't be too difficult.

mikedgit, 2022-10-10:
@warhammerkid, Yes the integration development is also plagued by the fact that they're constantly trying to tighten up how they do things but not break new integrations.  I'm working on another integration (spent most of today on it actually), and it's a little painful.

Depending on what your dev environment is, I found that [development in their remote containers](https://developers.home-assistant.io/docs/development_environment) is best for initial setup.  They offer [scaffolding scripts](https://developers.home-assistant.io/docs/creating_component_index) that are FAIRLY good at populating things.

If you didn't set up with those, it may actually be useful to just do the scaffolding like you're setting up a new integration, then copy over the stuff you did.  If you already did that, then you're already beyond my help and I should be asking you for advice on my integration ;).

If you have a custom_component public repo you want to check in and collaborate on, I'd be happy to test and help - I have a RPi4 Home Assistant instance with an AC300 + 2*B300s within easy bluetooth range standing by!

twister36, 2022-10-10:
Count me in too for testing. I also have a Raspberry Pi 4 with HA running and dual AC300's in split-phase mode and also dual AC500's in split-phase mode.

fluchderpixel, 2022-10-10:
When the time comes, I can also test, Raspi 4 with HA and AC200MAX...

Thank you very much for the work!

twister36, 2022-10-10:
@warhammerkid what version of HA are you running? On what platform?

warhammerkid, 2022-10-13:
> what version of HA are you running? On what platform?

@twister36 The full OS image on a Raspberry PI. I'm currently running HAOS 9.0 and core 2022.10.1.

warhammerkid, 2022-10-13:
@mikedgit Yeah, I read through their docs and instead of following their recommendations basically just started throwing code together to see how far I could get. Looking again at the scaffold templates and their example projects, I'm realizing that there's some useful information in the comments that I'll need to take a look at this weekend. I pushed up [what I have](https://github.com/warhammerkid/bluetti_ha), but it's basically nothing at the moment.

SSMCD, 2022-11-02:
@warhammerkid FWIW, I bundled this into a Home Assistant add-on to make the install a little easier for now:
https://github.com/SSMCD/bluetti2mqtt

I think a custom integration is still the cleanest way to the get the data into Home Assistant.

This got me up and running with my AC500 so I can start poking around the logs.

Running HA on a NUC with native Bluetooth.  Hasn't been tested on other platforms yet.

[image]

twister36, 2022-11-02:
@SSMCD Have you tried this with multiple MAC Addresses for two or more Bluetti units?

SSMCD, 2022-11-02:
I have not - were you able to get multiple devices working within one instance before?  If so, I should be able to make some tweaks to make the addon work as well.

twister36, 2022-11-02:
@SSMCD - I haven't tried.... I find that the bluetooth on my Bluetti's lockup once a week and I need to reset the bluemqtt service. I find it easier to reset when I have a service to each bluetti unit.  Plus I have 4 units.

SSMCD, 2022-11-03:
That's better than what I'm seeing - I stop receiving MQTT messages after a couple hours.  Program stays running without errors but Bluetti screen shows Bluetooth state disconnected.   Suspect it's poor signal/interference.  I'd really like to get this working with WiFI as well.

warhammerkid, 2022-11-05:
> Program stays running without errors but Bluetti screen shows Bluetooth state disconnected.

> I find that the bluetooth on my Bluetti's lockup once a week and I need to reset the bluemqtt service

@SSMCD @twister36 I have updated the error handling code in #40, so it's possible that will help in tracking things down. If you are continuing to have issues without anything being logged after these changes, it might help to run the service in debug mode temporarily (although that greatly increases log output). If you are using systemd you'd want to make this change:

```
[Service]
...
Environment="DEBUG=true"
ExecStart=/standard/path/to/bluetti-mqtt
```

If this helps to give you an error message, feel free to create a new issue on Github for me to look into it.

twister36, 2022-11-06:
@warhammerkid I find that it's not usually the blue-mqtt service that fails. The Bluetooth on the AC300 unit locks up and I need to usually turn off Bluetooth or reboot the AC300 to get it to work again. I've noticed that if I set the interval to >= 10 seconds then I don't have many issues with the AC300 Bluetooth locking up.

SSMCD, 2022-11-06:
@warhammerkid Awesome, thanks -  Bluetooth now reconnects after the connection has been lost.

I added a toggle for debug mode in the addon configuration.

mbay0r, 2023-01-05:
I would also very much welcome it.

then I could disable the virtual machine for this application

Patrick762, 2023-07-30:
I think the documentation for HA Integrations improved a bit https://developers.home-assistant.io/docs/bluetooth.
The device page is already working https://github.com/Patrick762/hassio-bluetti-bt (Test at own risk, still work in progress).
Polling and sensors are next.
If I saw it correctly, this project could be used as dependency for the integration

Patrick762, 2023-07-31:
All sensors from this project should now be available in the integration (tested with EP600 and AC300)

MegaBarf, 2023-08-01:
Hi everyone,

one short question, maybe I missed something, but what is the difference from this to https://github.com/SSMCD/bluetti2mqtt ?
I'm already using this one for quite some time now.

Patrick762, 2023-08-14:
@MegaBarf bluetti2mqtt is this repository, but bundled as a home assistant addon (to install it into home assistant supervised / os, see [HA docs](https://www.home-assistant.io/installation#compare-installation-methods)).
An integration on the other hand can be installed in any home assistant and can also use [bluetooth proxies](https://esphome.github.io/bluetooth-proxies/)