warhammerkid/bluetti_mqtt #24: Get an Error on Asustor NAS

fluchderpixel, 2022-09-02:
Hello,

I will try run Bluetti-Mqtt in my Asustor NAS AS6604. Blutooth is with an Bluetooth-dongle. The installation is gone without Error. But when i start Bluetti-Mqtt --scan i get the following error:

> root@HT730-NASSERVER:/volume1/.@root # bluetti-mqtt --scan
Scanning....
Traceback (most recent call last):
  File "/opt/bin/bluetti-mqtt", line 8, in <module>
    sys.exit(main())
  File "/opt/lib/python3.10/site-packages/bluetti_mqtt/server_cli.py", line 123, in main
    cli.execute()
  File "/opt/lib/python3.10/site-packages/bluetti_mqtt/server_cli.py", line 62, in execute
    asyncio.run(scan_devices())
  File "/opt/lib/python3.10/asyncio/runners.py", line 44, in run
  File "/opt/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
  File "/opt/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/__init__.py", line 17, in scan_devices
    devices = await BleakScanner.discover()
  File "/opt/lib/python3.10/site-packages/bleak/backends/scanner.py", line 117, in discover
    async with cls(**kwargs) as scanner:
  File "/opt/lib/python3.10/site-packages/bleak/backends/scanner.py", line 97, in __aenter__
    await self.start()
  File "/opt/lib/python3.10/site-packages/bleak/backends/bluezdbus/scanner.py", line 129, in start    manager = await get_global_bluez_manager()
  File "/opt/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 783, in get_global_bluez_manager
    await instance.async_init()
  File "/opt/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 225, in async_init
    assert_reply(reply)
  File "/opt/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownMethod] Method "GetManagedObjects" with signature "" on interface "org.freedesktop.DBus.ObjectManager" doesn't exist

I'm not shure, is the Error from Bluetti-Mqtt or from my NAS?

Thanks for your help.

warhammerkid, 2022-09-03:
I'm not sure your NAS is supported for Bluetti-Mqtt. In short, the code I'm using to connect with Bluetooth depends on [BlueZ](http://www.bluez.org/) being installed, and it talks to it through `DBus`. I think the error message implies that your NAS has DBus, but probably doesn't have BlueZ installed? Doing some Google searches for "Asustor BlueZ" isn't turning up any useful matches, so it might not be support (or nobody who's done it has talked about it)?

I did some further digging and it seems like it might be possible that you could install [Entware](https://github.com/Entware/Entware/wiki) and use that to install the BlueZ packages, but that also might not work? The Asustor runs Linux, but a very stripped down version of it, so it's not necessarily going to be easy.

fluchderpixel, 2022-09-08:
Direct use of Bluetti-MQTT in Asustor's base system has not brought any success. The stripped down Linux is just too limited. I now have Linux running in the VirtualBox on the NAS, it works great. Thanks for the considerations.

closed 2022-09-08 by fluchderpixel

gerab, 2023-08-01:
@warhammerkid please add it to README. Should run on Ubuntu Server
> sudo apt install bluez

Thanks