warhammerkid/bluetti_mqtt #25: Lost connection?

fluchderpixel, 2022-09-08:
Hi,

I read in the forum (https://diysolarforum.com/threads/monitoring-bluetti-systems.37870/post-592765) that Bluetti-MQTT seems to lose the connection and the broker does not receive any new data. The process (Bluetti-MQTT runs on Ubuntu in a VirtualBox.) is still running, my guess is that the service loses the Bluetooth connection and cannot reconnect. I'm not using IOBrocker, I'm running Home Assistant with MosquitoMQTT, but I'm having the same problem. After restarting the Bluetti-MQTT process, the connection is back and current data is received. Unfortunately, I don't have any other log entries because I couldn't find any that explain the problem.
[image]

fluchderpixel, 2022-09-11:
I took a closer look at the weekend and did a little testing. Bluetooth reception is obviously poor. If the Bluetooth connection is lost, the following error message appears:

> WARNING:bleak.backends.bluezdbus.client:Failed to cancel connection (/org/bluez/hci0/dev_00_11_22_33_44_55): [Errno 32] Broken pipe
ERROR:asyncio:Future exception was never retrieved
future: <Future finished exception=BrokenPipeError(32, 'Broken pipe')>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/bluetti_mqtt/bluetooth/client.py", line 38, in run
    await self.client.connect()
  File "/usr/local/lib/python3.10/dist-packages/bleak/backends/bluezdbus/client.py", line 167, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.10/dist-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local
> During handling of the above exception, another exception occurred:
> Traceback (most recent call last):

>   File "/usr/local/lib/python3.10/dist-packages/bleak/backends/bluezdbus/client.py", line 188, in connect reply = await self._bus.call(

>   File "/usr/local/lib/python3.10/dist-packages/dbus_next/aio/message_bus.py", line 63, in write_callback self.offset += self.sock.send(self.buf[self.offset:])

> BrokenPipeError: [Errno 32] Broken pipe

In my view, it's not the code, so I'm closing the issue. I'll see if I can find a Bluetooth stick with a long range to improve the stability of the connection.

warhammerkid, 2022-09-11:
My goal is that it should automatically reconnect, and it looks like this is happening because I'm simply not handling this exception. I'm going to re-open this and address this when I have some time.

PR #29 (2022-10-02): Catch previously unhandled exceptions in bluetooth connect
This fixes #25
closed 2022-10-02 by warhammerkid