warhammerkid/bluetti_mqtt #32: Units of measure for AC200MAX with incorrect factors

ra-ma, 2022-10-16:
Hi, first of all great project, thanks!

Monitoring data for my AC200MAX following jumped into my eye:
- dc_input_voltage1 too small by factor 10
- dc_input_power1 too large by factor 10
- dc_input_current1 too large by factor 10 

I corrected it "the dirty way" in file mqtt_client.py and tested successfully:

```python
       if 'internal_dc_input_voltage' in msg.parsed:
            await client.publish(
                topic_prefix + 'dc_input_voltage1',
                payload=str(10*msg.parsed['internal_dc_input_voltage']).encode()
            )
        if 'internal_dc_input_power' in msg.parsed:
            await client.publish(
                topic_prefix + 'dc_input_power1',
                payload=str(msg.parsed['internal_dc_input_power']/10).encode()
            )
        if 'internal_dc_input_current' in msg.parsed:
            await client.publish(
                topic_prefix + 'dc_input_current1',
                payload=str(msg.parsed['internal_dc_input_current']/10).encode()
            )
```

warhammerkid, 2022-10-16:
This is pretty easy for me to fix by updating [ac200m.py](https://github.com/warhammerkid/bluetti_mqtt/blob/main/bluetti_mqtt/core/devices/ac200m.py) for the config for those fields. Would it be possible to ask for you to collect and send me some logs from your device while you have some solar input? I just want to have some reference data to confirm I have things set up properly. If not I can make the changes without it and I’ll just have you confirm it’s correct.

warhammerkid, 2022-10-17:
@ra-ma Turns out I already had some reference data from somebody else that I was able to use for confirmation. Thanks for the feedback - version 0.9.1 will have this fixed.

closed 2022-10-17 by warhammerkid

ra-ma, 2022-10-17:
Great. Thanks. Happy to check it in new version and also happy to support for whatever testing needed for AC200Max in future.

PR #33 (2022-10-17): Fix decimal conversion factors for DC inputs on the AC200Max
The AC200Max has slightly different decimal placements for the internal DC input fields than newer models. Going back through my reference data I have confirmed that these new conversion factors are correct. Fixes #32.