warhammerkid/bluetti_mqtt #37: Batterypack Detals

woody3333, 2022-11-01:
Hi there,

this is not really an issue, just the question about extended function. I'm using AC200MAX extended with two batterypacks. 
Currently there is only the total_battery_percent delivered to Homeassitant.
I just seen via MQTT explorer that the Battery details are already read out and available via MQTT:

pack_details2 = {"percent":58,"voltages":[3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37,3.37]}
pack_details3 = {"percent":53,"voltages":[3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38]}
pack_details1 = {"percent":35,"voltages":[3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38,3.38]}

I there a possibility to have the charging percentage of each Battery also available within Homeassistant?

Regards and greetings from Germany
Michael

PS: Great work so far and very useful tool for all HA and Bluetti useres.

slash5k1, 2022-11-12:
Created PR #41 to add support for battery pack details. - just needs to be tested :)

closed 2022-11-20 by warhammerkid

woody3333, 2022-11-23:
Just tested the new release 0.11.0 and it is working as expected. Well done. Thank you.

PR #45 (2022-11-20): Most fields should now be exported to Home Assistant
Rather than continue to get individual requests for specific fields to be added for Home Assistant, I'm going ahead and just adding (almost) everything. Battery cell voltages are currently not being exported even under the advanced flag, but I might change my mind about this if I get requests.

Fixes #37 & #42