warhammerkid/bluetti_mqtt #42: Are the new EB3A entities supported in Home Assistant?

burpen, 2022-11-12:
I have an EB3A that I've been using since version 0.9.0 and it has the following entities in Home Assistant:

[image]

In the 0.9.2 release, the EB3A config was updated to include additional functionality (eco mode, charging mode, etc.). The problem I have is that even after upgrading to 0.9.2 and beyond (currently running 0.10.0), Home Assistant still only lists the entities shown in the screenshot above. I do see the additional entities if I examine the bluetti topic using MQTT Explorer:

[image]

I assume HA is somehow stuck with a stale autodiscovery manifest, and I'm not sure how to force rediscovery to happen. I have tried stopping bluetti_mqtt, deleting the device in HA, then restarting bluetti_mqtt, to no change. Perhaps this is a Home Assistant problem more than a bluetti_mqtt problem.

On a related note, is the readme's list of supported sensors and switches up to date? I'm not sure if I should take it to mean that anything beyond what's listed there will require some additional development. If that's the case then what I'm seeing may be expected behavior.

slash5k1, 2022-11-13:
The home assistant discovery topic is being published by the code in mqtt_client.py under the _send_discovery_message function which covers:
- ac_input_power
- dc_input_power
- ac_output_power
- dc_output_power
- total_battery_percent
- ac_output_on
- dc_output_on; and
- led_mode (if the device support led_mode)

Reads like you want the discovery to also include
 - eco_on
 - eco_shutdown
 - charging_mode; and
 - power_lifting_on

You could add these additional items to mqtt_client.py but maybe it would be better to only have the core auto discovery topics in this file and move specific device discovery to the device file?

burpen, 2022-11-18:
> Reads like you want the discovery to also include ...

Yes, you're spot on about what I was looking for. I haven't had the chance to try out #45, but it looks like it will address everything.

closed 2022-11-20 by warhammerkid