warhammerkid/bluetti_mqtt #43: Unable to power_off AC200M

slash5k1, 2022-11-13:
In the mobile app I have the option of turning off the unit completely and yet there is no implementation for it in the AC200M device file.

Looking through the code I can see this control defined for eb3a:
`self.struct.add_bool_field('power_off', 0x0B, 0xF4)`

Added this to the ac200m device file and can confirm when I publish "power_off" to the command topic the unit turns off! :)

For completeness can I suggest that each device is updated to include this control command. - assuming all devices support the power_off command?

warhammerkid, 2022-11-13:
My AC300 definitely does not support this (just double checked), and the mobile app has never offered the option. My guess is that the EP500 and EP500P also do not support it, since they're very similar to the AC300 in behavior. I think the best thing to do is add it just for the AC200M for now, and if anyone asks for it for the AC500 or EP500 I can walk them through the process to test it.

slash5k1, 2022-11-14:
Wow - so this is what I see in my mobile app for the AC200M:

[image]

Then when I press power off I am prompted with this:

[image]

I'll create the PR to update the AC200M

closed 2022-11-20 by warhammerkid