warhammerkid/bluetti_mqtt #47: It's MODBUS!

warhammerkid, 2022-11-20:
I don't know why it took me so long to realize this, but the command/response API is just MODBUS over Bluetooth (and MODBUS over MQTT for the cloud support over WiFi). The internal API should be reworked to use standard terminology, as well as proper exception handling.

Marko298, 2022-11-23:
Does it mean we can connect somewhere on the board to modbus?

warhammerkid, 2022-11-25:
@Marko298 Sorry, I don’t really know the answer to that. Taking my AC300 apart is a bit outside my comfort zone.

mark-b-ab, 2022-11-25:
It has communication interface exposed right below the DC input

tinkercnc, 2023-04-11:
@mark-b-ab OK, and do you know the pinout?

Marko298, 2023-04-12:
> @mark-b-ab OK, and do you know the pinout?

Nope, I can try to probe with logical analyzer. But I will be able to do it only in a week or so

tinkercnc, 2023-04-12:
I'm certainly looking forward. :)

Patrick762, 2024-02-03:
If I understood the device data correctly, there should be only 2 variants of address ranges. The only question is, which device uses which variant. With that info we could make this software much simpler to include new models of powerstations