warhammerkid/bluetti_mqtt #54: Add ability too see cell voltage.

TheUser, 2023-01-05:
I think it would be a great addition to be able to see and track individual cell voltages,

On my ep500p i can see this in the logs

{'pack_num_max': 1, 'pack_voltage': Decimal('53.8'), 'pack_battery_percent': 96, 'pack_num': 1, 'cell_voltages': [Decimal('3.35'), Decimal('3.36'), Decimal('3.35'), Decimal('3.35'), Decimal('3.36'), Decimal('3.35'), Decimal('3.35'), Decimal('3.35'), Decimal('3.35'), Decimal('3.35'), Decimal('3.35'), Decimal('3.36'), Decimal('3.36'), Decimal('3.35'), Decimal('3.35'), Decimal('3.35')]}

warhammerkid, 2023-01-15:
This information is published to `bluetti/state/[DEVICE NAME]/pack_details1` as JSON:

```js
{"percent":96,"voltages":[3.35,3.36,3.35 /* ... */]}
```
I'm not automatically configuring this in Home Assistant because it doesn't seem useful to most users, but you can add your own sensor if you'd like this information.

MegaBarf, 2023-02-14:
please add this feature as I don't know how to do that. 
I would like to keep track of the pack voltage, at least the decimal value as the battery pack keeps declining in % when youu don't have any inputs or outputs. The % declines about 1% in 3 hours. I would like to see if this is an error from the BMS or if the voltage is declining in reality.

MegaBarf, 2023-02-14:
Would be nice to get each cell voltage as one sensor and one sensor with full voltage (add up all voltages)

At my EP500P it looks like that:
{"percent":79,"voltages":[3.33,3.33,3.33,3.34,3.33,3.33,3.33,3.33,3.33,3.33,3.33,3.33,3.33,3.33,3.33,3.33]}

"TheUser" had something diffent where I have percent":79,"voltages he has 'pack_voltage': Decimal('53.8')

the full voltage would be nice

TheUser, 2023-02-15:
@MegaBarf 

I manually made 16 sensors in HomeAssistant using this code, there might be a simpler way of doing it but it works.

mqtt:
  sensor:
    - name: "Thor Cell 1 Voltage"
      state_topic: "bluetti/state/EP500P-2232000072818/pack_details1"
      unit_of_measurement: "V"
      value_template: "{{ value_json.voltages[0] }}"   
    - name: "Thor Cell 2 Voltage"
      state_topic: "bluetti/state/EP500P-2232000072818/pack_details1"
      unit_of_measurement: "V"
      value_template: "{{ value_json.voltages[1] }}" 
      
And so on until you get all 16 cells, after that i used a helper to group all 16 cells in to a new sensor that shows the total pack voltage.

MegaBarf, 2023-02-16:
@TheUser Thank you very much. It worked. For the Noobs like me I'll post what to do:

-add "mqtt: !include mqtt.yaml" to the configuration.yaml
-add a new file (mqtt.yaml) by using the file editor, or upload the one i attached (don't forget to rename to mqtt.yaml first).
[mqtt.yaml.txt](https://github.com/warhammerkid/bluetti_mqtt/files/10756400/mqtt.yaml.txt)
-open the mqtt.yaml and change your EP500 Serial number at state_topic: "bluetti/state/EP500P-2214000010???/pack_details1" on every row.

This will get you 16 sensors for each battery voltage.

- go to settings / devices&services / helpers
- Create helper
- chose "combine the state of serveral sensors"
- define a name for your new sensor (like Bluetti Battery Pack Voltage)
- add all Cells, in my case "Bluetti Cell 1 Voltage" - "Bluetti Cell 16 Voltage"
- set "statistic characteristics" to sum and submit.

Done!

MegaBarf, 2023-02-16:
@warhammerkid 

would still be nice to have it included by default (at least the cell voltages).

Thank you very much for your work.

ibuvery, 2023-02-17:
I've found that cell voltages (pack voltage) is indeed very important to overall health of the battery pack in "resting" state.  Also, the individual cell voltage deviation from each other (during charge and discharge) could indicate premature individual cell deterioration.   I've found it best to leave the battery pack at specific voltages to reduce % drops overnight  (it's different for each pack).

I've been capturing this outside of HA just because this ability was missing in HA, and the info above is useful.  However, it would be great to include the pack/cell voltages within the project in a more consumable format so that further automation can be done easily.  If others are not using the voltages today they probably would/should use it in the future as the data is more readily available and more people analyze the data.

Thank you @TheUser for opening the issue and @MegaBarf for the comments.  Your contributions @warhammerkid are greatly appreciated.

steveheinsch, 2023-03-02:
@TheUser I believe you got that from my post on Home Assistant? Here's the complete thread where I was asking the HA crowd if creating the individual cell voltages from the array could be automated instead of having to create 16 entries (per battery) manually.

https://community.home-assistant.io/t/display-array-of-cell-voltages-from-mqtt/523450

ibuvery, 2023-06-03:
The latest updates added total pack voltage, this is good news and a step in the right direction.  While this is helpful, I still found I needed to sum the individual cell voltages.  The higher precision allows me to calculate cell state (Pack V/16) and automate charging so that nightly drops don't bring down battery charge percentages past  (-1%)/hour.   

It would be even more useful to have two digit (hundredths) precision for the pack voltage.  
Thank you for all the contributions on this project!

warhammerkid, 2023-06-04:
> It would be even more useful to have two digit (hundredths) precision for the pack voltage.

@ibuvery That pack voltage value I'm reporting is the value directly from the device, which is only provided by the firmware with a single digit of precision. My aim is simply to report whatever data Bluetti makes available in as close a form as that data was originally available, and leave interpretation to users. I'm assuming that Bluetti has a different sensor for the pack voltage than they have for the cell voltages, since it's not exactly the sum rounded to the tenths place.