warhammerkid/bluetti_mqtt #56: Fix entities for HAs Energy Management

mbay0r, 2023-01-05:
I can't find/add the entity DC Input Power and AC Input Power into the [HAs Energy Management](https://www.home-assistant.io/blog/2021/08/04/home-energy-management/).

It is possible to add here maybe two more entities following this guide or customize it?

https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

thank you very much!

twister36, 2023-01-10:
In order to add metrics to your HA Energy Dashboard it must be in a format that accepted. You'll need to create a sensor in your configuration.yaml file like such:

sensor:
  - platform: integration
    unique_id: ac500_a_total_dc_input_power
    source: sensor.ac500_dc_input_power
    name: "AC500-A TOTAL DC INPUT POWER"
    unit_prefix: k
    unit_time: h
    
    
   Then you can add the new entity "AC500-A TOTAL DC INPUT POWER" in your energy dashboard. 
   
   Note, this isn't a Bluetti-MQTT issue.. this a question for the HA community on how to take the data coming into HA and format it so it works within in HA as you wish.

mbay0r, 2023-01-11:
Thanks @twister36 !

Just so I understand it better:

But with other products that I have in use, it works directly.
The addon creates the entity, doesn't it?
 So it could also create them directly correctly, so I don't have to add a new sensor?

warhammerkid, 2023-01-15:
> So it could also create them directly correctly, so I don't have to add a new sensor?

@mbay0r The Bluetti devices don't have kWh data, so the only way to get that is to approximate it by taking various wattage readings over time and estimating how many kWh have been used. If done incorrectly, this estimate can be off by quite a bit. I haven't had any opportunity to understand how well the [integration](https://www.home-assistant.io/integrations/integration/#energy) sensor works for Home Assistant, so I currently don't auto-configure it.

I know that long-term kWh statistics is something that users want, and it is on my list to support.