warhammerkid/bluetti_mqtt #59: Does not find AC200MAX on raspberry

nuagevert59, 2023-02-04:
Hello

I installed the bluetti_mqtt in order to connect my AC200MAX to Jeedom

Installation ran well using `sudo pip3 install bluetti_mqtt`

then when i try to connect using `bluetti-mqtt --scan` , it fails `0 devices found`
here is example :
`pi-/home/pi> hcitool dev`
`Devices:`
`        hci0    00:01:95:7C:01:18`
`pi-/home/pi> lsusb | grep -i bluetooth`
`Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`
`pi-/home/pi> bluetti-mqtt --scan`
`Scanning....`
`0 devices found - something probably went wrong
`

BT Dongle is well known, at RPI can see the devices, and the AC200MAX is in the list :
[image]

BT is ok on the AC200MAX  : i can connect with android app
the AC200MAX  is close to the RPI
  
if you have any idea of what's wrong with me, 
thank you for your help

Here is my stuff :
Raspberry pi 3B running rasbian 11 bullseye
Bluetooth dongle : SENA UD100-G03
AC200MAX : 
   ARM version : v4008,03
   DSP version : v4005,07
   BMS version : v1017,02

Duker69, 2023-02-12:
I have the same problem with AC200MAX, Raspberry pi 4B, Debian
[image]

warhammerkid, 2023-02-12:
Can you do me a favor and run `DEBUG=true bluetti-mqtt --scan`? If everything is working it should print out a bunch of lines that start with `received D-Bus signal: org.freedesktop.DBus.`. If it doesn't then there might be something wrong with Bluez or DBUS? Also, if you could post the results of running `pip3 list` that would help me check if it's some dependency version that is the problem.

Duker69, 2023-02-12:
[image]

nuagevert59, 2023-03-16:
and the pip3 list 
`pi@pi2:/home/pi> pip3 list
Package               Version
--------------------- --------------
android-backup        0.2.0
appdirs               1.4.4
asn1crypto            1.4.0
async-timeout         4.0.2
asyncio-mqtt          0.16.1
attrs                 22.2.0
bleak                 0.19.5
bluepy                1.3.0
bluetti-mqtt          0.11.1
certifi               2022.12.7
cffi                  1.15.1
chardet               4.0.0
charset-normalizer    3.0.1
click                 8.1.3
construct             2.10.68
crcmod                1.7
croniter              1.3.8
cryptography          39.0.1
dbus-fast             1.84.2
dbus-python           1.2.16
defusedxml            0.7.1
distro                1.5.0
distro-info           1.0
docutils              0.16
entrypoints           0.3
enum-compat           0.0.3
enum34                1.1.10
fail2ban              0.11.2
future                0.18.3
gpg                   1.14.0-unknown
idna                  3.4
ifaddr                0.2.0
iotop                 0.6
jeepney               0.6.0
keyring               22.0.1
keyrings.alt          4.0.2
micloud               0.6
netifaces             0.11.0
numpy                 1.24.2
olefile               0.46
paho-mqtt             1.6.1
Pillow                8.1.2
pip                   23.0.1
PyBluez               0.23
pycparser             2.21
pycrypto              2.6.1
pycryptodome          3.17
pycryptodomex         3.9.7
pycurl                7.43.0.6
Pygments              2.7.1
PyGObject             3.38.0
pyinotify             0.9.6
pyserial              3.5b0
python-apt            2.2.1
python-dateutil       2.8.2
python-miio           0.5.12
pytz                  2022.7.1
pytz-deprecation-shim 0.1.0.post0
pyudev                0.22.0
pyxdg                 0.27
PyYAML                6.0
requests              2.28.2
roman                 2.0.0
SecretStorage         3.3.1
setuptools            67.4.0
six                   1.16.0
ssh-import-id         5.10
systemd-python        234
tqdm                  4.64.1
typing_extensions     4.5.0
tzdata                2022.7
tzlocal               4.2
unattended-upgrades   0.1
urllib3               1.26.14
wheel                 0.34.2
xdg                   5
yeelight              0.7.10
zeroconf              0.47.3
`

thank you

nuagevert59, 2023-04-12:
After a fresh install of Raspberry (bullesey) , it works well ❤️‍🩹

closed 2023-04-12 by nuagevert59