warhammerkid/bluetti_mqtt #61: Question: Is it possibe to limit AC charge input for AC200MAX?

itn3rd77, 2023-02-09:
Hi folks!

I am plaing to buy a AC200MAX. Is anyone here using an AC200MAX and can tell me if it is posible to control the AC charge input e.g. limit to 300 Watt?

Thanks and best regards!
Ingo

mbay0r, 2023-02-09:
No, this will not work, because there is no function for this in the APP.

itn3rd77, 2023-02-09:
> No, this will not work, because there is no function for this in the APP.

Thanks for clarifying that.

closed 2023-02-09 by itn3rd77

gerab, 2023-08-01:
@itn3rd77, you can create a simple service and set the Tasmota MQTT relay on your AC200 charger. That's what I've done. It works like a charm.

virtualj, 2023-12-11:
> @itn3rd77, you can create a simple service and set the Tasmota MQTT relay on your AC200 charger. That's what I've done. It works like a charm.

Hi, can you explain better what did you do? Are you able to limit the charging power from the AC200 charger?

gerab, 2023-12-11:
> > @itn3rd77, you can create a simple service and set the Tasmota MQTT relay on your AC200 charger. That's what I've done. It works like a charm.
> 
> Hi @gerab, can you explain better what did you do? Are you able to limit the charging power from the AC200 charger?

Hi. Yes, I wrote a .net service, which monitors AC200 battery levels and toggles the AC charger when the level is below-configured minimum. It charges it until the battery level reaches the expected percent. If you're familiar with .net and c# I can share the source code with you.

virtualj, 2023-12-11:
@gerab , this is not what @itn3rd77 asked. I do what you did with Home Assistant: I enable AC charger (with smart plug) only when battery is below 15%. But in this way the charger charge always @500W. What is usefull to do is to limit the charge power to less, for example 200W, because I have another FTV on grid and I can have only 200W surplus to put in the storage