warhammerkid/bluetti_mqtt #63: Cell and System temperatures

patilap, 2023-02-18:
Hey, 

this is really great work. I use my AC200MAX in my camper and I finally have a great solution for my energy management which is quite important when you stay offroad without a constant electricity source.

Last year I ran into some temperature issues that even led to a replacement of my device because it ran into ntc error all the time.

Therefore I am a little bit concerned about the temperature sensor readings of the device and want to track them.

I was wondering if there are more values delivered from the device via Bluetooth that haven´t been integrated yet? Especially temperatures.

In this video https://www.youtube.com/watch?v=hYgJl9m6eZg you can see temperature values and individual cell voltages in the Bluetti Device UI.

Thank you in advance! Greetings from Germany

steveheinsch, 2023-03-02:
@patilap The cell values are available, but you have to do a bit of work to get them into Home Assistant. Currently the voltages for each cell in a pack are provided as an array and not single entities. See this thread for some workarounds: https://github.com/warhammerkid/bluetti_mqtt/issues/54

patilap, 2023-03-03:
@steveheinsch  yea thanks, I know that and have that already installed. In addition I am looking for System or Cell temperatures.

warhammerkid, 2023-03-12:
The AC200M does not appear to have temperature sensors available over Bluetooth. I know they have them since they’re displayed on the screen, but the firmware only exposes some information over bluetooth.

SSMCD, 2023-03-30:
Looks like pack temperature (°F) is stored at 0x5F on page 00 for the AC500.