warhammerkid/bluetti_mqtt #65: EOFError crashes bluetti-mqtt

iandees, 2023-03-01:
I'm running bluetti-mqtt to connect to my AC500. Every once and a while I get this stack trace that crashes the app:

```python
2023-03-01 08:48:49 ERROR    Error connecting to device xx:xx:xx:xx:xx:xx:
Traceback (most recent call last):
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bluetti_mqtt/bluetooth/client.py", line 40, in run
    await self.client.connect()
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 172, in connect
    reply = await self._bus.call(
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/aio/message_bus.py", line 370, in call
    await future
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/aio/message_reader.py", line 24, in _message_reader
    message = unmarshaller._unmarshall()
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 641, in _unmarshall
    self._read_header()
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 524, in _read_header
    self._read_to_pos(HEADER_SIGNATURE_SIZE)
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 276, in _read_to_pos
    raise EOFError()
EOFError
```

warhammerkid, 2023-03-19:
It _should_ automatically retry even if it prints out this exception. When you say that it crashes the app what do you mean?

iandees, 2023-03-19:
`bluetti_mqtt` exits. Here's the full log from the most recent event:

```
2023-03-02 00:21:24 ERROR    Crashing with uncaught exception:
Traceback (most recent call last):
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bluetti_mqtt/device_handler.py", line 31, in run
    await asyncio.gather(*(polling_tasks + pack_polling_tasks + [manager_task]))
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bluetti_mqtt/bluetooth/manager.py", line 19, in run
    await asyncio.gather(*[c.run() for c in self.clients.values()])
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bluetti_mqtt/bluetooth/client.py", line 53, in run
    await self._perform_commands(self.client)
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bluetti_mqtt/bluetooth/client.py", line 75, in _perform_commands
    await client.write_gatt_char(
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bleak/__init__.py", line 593, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 788, in write_gatt_char
    reply = await self._bus.call(
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/aio/message_bus.py", line 370, in call
    await future
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/aio/message_reader.py", line 24, in _message_reader
    message = unmarshaller._unmarshall()
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 641, in _unmarshall
    self._read_header()
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 524, in _read_header
    self._read_to_pos(HEADER_SIGNATURE_SIZE)
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 276, in _read_to_pos
    raise EOFError()
EOFError
2023-03-02 00:21:24 INFO     Shutting down...
```

warhammerkid, 2023-03-20:
Thanks, that's helpful. I'll need to dig a bit further into why this is happening, but it shouldn't be too difficult to handle this.

PR #76 (2023-04-29): Lazy Bluetooth Connect
This change allows bluetti-mqtt to be started with device addresses that may not currently be in range. Once the device comes in range it will automatically be connected to, after discovering the device type. I also switched to a state machine for the bluetooth client class to clean things up a bit.

Fixes #60, fixes #65
closed 2023-04-29 by warhammerkid