warhammerkid/bluetti_mqtt #66: request for EP600

LX1974, 2023-03-02:
Hello Warhammerkid,

I have a EP600 with 2 EP500, can you this device?

thanks a lot!

Alex

lkm220929, 2023-03-15:
Hi,

I am also very interested in the ability to connect to an EP600. 

Kind regards
Martin

gedongi, 2023-03-19:
Hallo Alex, Martin I could give you some support. I already grabing some datas from the EP600.

LX1974, 2023-03-19:
Hello gedongi, yes please, let me know how to grab some data.

lkm220929, 2023-03-19:
Hi,thank you very much, I really appreciate your help with this. Please let me know if Ican test something or if I can be of any other help.Kind regards.Martin

LX1974, 2023-03-19:
Hi Martin, thanks a lot!

How can I grab some EP600 data?

gedongi, 2023-03-19:
Here's a link to another Github user I've been active with. There I wrote a PR and made comments. Please read first. If you have any questions after that, please comment.
I'm also looking for people which are interested in finding out  grab more datas from the EP600. If you are interested please leave a comment...may be we could join together for testing...

https://github.com/mariolukas/Bluetti_ESP32_Bridge/issues/40

Cheers

lkm220929, 2023-03-19:
Hi,I am owning an EP600, but don‘t really know how I could help to grab some data, I‘m just a „stupid“ consumer. Would need some instructions on how to get to the data.

warhammerkid, 2023-03-25:
The EP600 is completely different from all their other devices. I have added the `bluetti-discovery` command to the latest version to help start the process of adding support. Please run the command and attach the resulting log file. If you have solar panels hooked up and an AC load then noting down those values and including them with the resulting log file will help me to figure out the correct way to parse those out. From there it's just the slow process of figuring it all out - this is what that looks like for the [AC300](https://doc.chromedshark.com/bluetti/ac300.html) for example.

> For supporting new devices, the `bluetti-discovery` command is provided. It will scan from `0x0000` to `0x30ff` assuming MODBUS-over-Bluetooth. This will take a while and requires that the scanned device be in close Bluetooth range for optimal performance.
>
> `$ bluetti-discovery --scan`
> `Found AC3001234567890123: address 00:11:22:33:44:55`
> `$ bluetti-discovery --log the-log-file.log 00:11:22:33:44:55`

lkm220929, 2023-03-27:
Do you still require the result of the discovery scan ? If yes I can pull one from my device. Is there a way to send the results directly to you ? Not sure if I want to make it publicly available.

LX1974, 2023-04-02:
Log:
[log-ep600.log.zip](https://github.com/warhammerkid/bluetti_mqtt/files/11131462/log-ep600.log.zip)

warhammerkid, 2023-04-02:
@lkm220929 You're welcome to email it to me at perl.programmer@gmail.com.

@LX1974 I deleted your comment because it looks like it probably has data in it that shouldn't be public. I should have provided an email earlier.

warhammerkid, 2023-04-02:
The first step of the discovery process is to figure out where data can be requested, and the result of that using the data @LX1974 provided is something [like this](https://doc.chromedshark.com/bluetti/ep600.html). I have removed your wifi access point and password, as well as changing your serial number. I _think_ that should be everything that needs to be removed.

This is a static snapshot of the data at the time you ran the discovery. The next step is to add basic support for logging data from the EP600 so we can start changing things and seeing what changes in the data. Some things are easy enough to figure out from a static snapshot (like the device type or serial number), but on/off switch settings are almost impossible to figure out that way. However, with the dynamic log if you change a specific setting at a specific time, then we should be able to figure out what address corresponds to that setting based on what changes at that time.

lkm220929, 2023-04-05:
@warhammerkid: I have sent you some information via email. Can you confirm that you have received it ?

warhammerkid, 2023-04-05:
@lkm220929 I saw your email but haven’t yet had a chance to review it.

lkm220929, 2023-04-06:
Thanks for the confirmation. Review it, whenever you have a chance.

lkm220929, 2023-05-01:
Hi warhammekid,

just wanted to check if you had the chance to look into the EP600 integration yet ?

warhammerkid, 2023-05-13:
The next steps for EP600 support are to use the `bluetti-logger` data to reverse engineer specific fields.

For most of our needs there are four different kinds of fields and the process for reverse engineering varies for each type:
- Numeric fields
- Boolean fields (on/off or yes/no)
- Enum fields (options or modes)
- String fields (like a device type or password)

For numeric fields like the battery percent or solar input, the process is to run the logger and write down the values you see in the app over wifi, and the time when you saw those values. For rapidly changing values a range can be noted down. For slowly changing values it may be necessary to collect multiple logs at various values to make it easier to figure out where the field is. You can write down multiple values at a given time so that you don't need to run the logger for as long.

For boolean and enum fields, the process is to run the logger and to change the field in the app, noting down the value and the time when you changed it. You should wait at least a minute between changes, and for boolean fields it can help with identification to change the value several times.

I'm pretty sure I identified all the string fields but if there's a specific field you're looking for please let me know and I can help look for it.

---------

A faster option is to use a bluetooth packet logger (like an ESP32 flashed with nRF connect app or the Android / iOS developer tools) instead of `bluetti-logger`, and to use a mobile device to connect directly to the EP600 over Bluetooth. **I am not a lawyer so before you do this you should check that this is legal in the country you reside in.** With this method you don't need to wait a whole minute after changing a settable field - simply note down the time when the value was changed and wait a few seconds before changing it again. This method usually helps speed up the search for sensor values as well, since the app only loads the data it needs for the specific screen you are on, narrowing down the range where that value could be located.

lkm220929, 2023-05-15:
Hi, did you have the chance to review the email I sent in April ? Was that information of any help for the reverse engineering ?

lkm220929, 2023-06-30:
I couldn’t figure out how to convert the BLUETTI-logger output into this format  the BLUETTI-logger output into the html format you shared earlier (https://doc.chromedshark.com/bluetti/ep600.html) ? Reverse Engineering would be easier if I could convert my logger data myself.

Patrick762, 2023-08-02:
I found some values which look like they belong to the second battery:

- 0x49B8 - 0x49BD battery_type
- 0x49BE - 0x49C1 battery_serial_number
- 0x4A02 - 0x4A03 bcu_version
- 0x4A05 - 0x4A06 bmu_version
- 0x4A08 - 0x4A09 safety_module_version
- 0x4A0B - 0x4A0C high_voltage_module_version

I also found the mac address at 0x2B13 - 0x2B15

@warhammerkid Did you use a script to generate the html content?
How can I get a version number from the raw hex data (like the one in the html file)?

Patrick762, 2023-08-03:
It seems like at some point reading the registers isn't working (f. ex. 14085, 14091, 14097) which would be needed to add more fields from the EP600

Patrick762, 2023-08-03:
Here are some more fields I found:

- 0x04CD adl400_ac_input_voltage_phase1
- 0x04D5 adl400_ac_input_voltage_phase2
- 0x04DD adl400_ac_input_voltage_phase3

Patrick762, 2023-08-03:
Here are some new fields I found in my discovery data (I deleted my previous comments because they contained wrong information):

```
self.struct.add_decimal_field("adl400_ac_input_voltage_phase1", 1229, 1)
self.struct.add_decimal_field("adl400_ac_input_voltage_phase2", 1237, 1)
self.struct.add_decimal_field("adl400_ac_input_voltage_phase3", 1245, 1)
self.struct.add_decimal_field("ac_input_frequency", 1300, 1)
self.struct.add_decimal_field("ac_input_voltage_phase1", 1314, 1)
self.struct.add_decimal_field("ac_input_voltage_phase2", 1320, 1)
self.struct.add_decimal_field("ac_input_voltage_phase3", 1326, 1)
self.struct.add_decimal_field("ac_output_frequency", 1500, 1)
self.struct.add_decimal_field("ac_output_voltage_phase1", 1511, 1)
self.struct.add_decimal_field("ac_output_voltage_phase2", 1518, 1)
self.struct.add_decimal_field("ac_output_voltage_phase3", 1525, 1)
```

StefanMoellenhoff, 2023-08-10:
Hey, I'm very interested in this as well, unfortunately I'm rather new to HA, but I have an EP600 and am happy to try out things and deliver data!

Patrick762, 2023-09-24:
There's now a PR (#87) for this issue thanks to @KM011092

andrewagner86, 2023-11-10:
@warhammerkid maybe you can merge the PR for EP600?

smai86, 2023-12-12:
> @warhammerkid maybe you can merge the PR for EP600?

Yeah that would be awesome! Is it possible to install the Version from @KM011092 without merging from @warhammerkid ?

andrewagner86, 2023-12-12:
i dont know how you are trying to use it, but of course you can use the version from @KM011092 standalone. but there are some obvious bugs in it, so its not the 100% solution for ep600

Patrick762, 2023-12-12:
In theory you could install it by running the following commands (this will overwrite the existing installation):

```sh
git clone https://github.com/KM011092/bluetti_mqtt.git
cd bluetti_mqtt/
pip install .
```

smai86, 2023-12-12:
> In theory you could install it by running the following commands (this will overwrite the existing installation):
> 
> ```shell
> git clone https://github.com/KM011092/bluetti_mqtt.git
> cd bluetti_mqtt/
> pip install .
> ```

Thanks @Patrick762 I think I would try this tomorrow! I also saw your HA intregration. Can u tell me if it would work with an ESP32 as BT Proxy (over ESP Home)? My HA server is to far away from my EP600 to get a BT connection with a BT dongle. So I saw the possibility with a BT Proxy over ESP32. That would be the easiest way for me to get the data from my EP600

Patrick762, 2023-12-13:
@smai86 you can use the integration it with an eshome bt proxy

smai86, 2023-12-13:
> @smai86 you can use the integration it with an eshome bt proxy

Yes I tested it today and it worked like a charme! Thanks a lot!!!
But i got one little problem and maybe you got a solution for me. The BT connects and disconnect every 20 sec. So I got the data only updated every 20 sec... 

You can see it here in my log file. Maybe you got a solution for me, that the BT connection keeps connected that the data gets updated faster...

09:50:31][I][bluetooth_proxy:282]: [0] [B4:8A:0A:5B:AB:A5] Connecting v3 without cache
[09:50:31][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[09:50:31][I][esp32_ble_client:069]: [0] [B4:8A:0A:5B:AB:A5] 0x00 Attempting BLE connection
[09:50:31][I][esp32_ble_client:201]: [0] [B4:8A:0A:5B:AB:A5] Connected
[09:50:31][D][esp32_ble_tracker:266]: Starting scan...
[09:50:32][I][esp32_ble_client:086]: [0] [B4:8A:0A:5B:AB:A5] Disconnecting.
[09:50:32][D][esp-idf:000]: W (32718) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[09:50:32][D][esp-idf:000]: W (32725) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[09:50:32][D][esp-idf:000]: W (32728) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[09:50:32][D][esp-idf:000]: W (32731) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[09:50:32][D][esp-idf:000]: W (32770) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

[09:50:52][I][bluetooth_proxy:282]: [0] [B4:8A:0A:5B:AB:A5] Connecting v3 without cache
[09:50:52][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[09:50:52][I][esp32_ble_client:069]: [0] [B4:8A:0A:5B:AB:A5] 0x00 Attempting BLE connection
[09:50:53][I][esp32_ble_client:201]: [0] [B4:8A:0A:5B:AB:A5] Connected
[09:50:53][D][esp32_ble_tracker:266]: Starting scan...
[09:50:53][I][esp32_ble_client:086]: [0] [B4:8A:0A:5B:AB:A5] Disconnecting.
[09:50:53][D][esp-idf:000]: W (53798) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[09:50:53][D][esp-idf:000]: W (53805) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[09:50:53][D][esp-idf:000]: W (53808) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[09:50:53][D][esp-idf:000]: W (53812) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[09:50:53][D][esp-idf:000]: W (53845) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

[09:51:13][I][bluetooth_proxy:282]: [0] [B4:8A:0A:5B:AB:A5] Connecting v3 without cache
[09:51:13][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[09:51:13][I][esp32_ble_client:069]: [0] [B4:8A:0A:5B:AB:A5] 0x00 Attempting BLE connection
[09:51:14][I][esp32_ble_client:201]: [0] [B4:8A:0A:5B:AB:A5] Connected
[09:51:14][D][esp32_ble_tracker:266]: Starting scan...
[09:51:15][I][esp32_ble_client:086]: [0] [B4:8A:0A:5B:AB:A5] Disconnecting.

KM011092, 2024-01-28:
Just updated my EP600 to the newest Software releases, with the result, that the integration does not work anymore. I could not yet find out why. So if you don’t have a need to update your EP600 - Don’t do it !!!

gedongi, 2024-01-28:
I had the same issue after last IOT Update. The Problem was that Bluetti changed the Station name in the IOT device, so bluetooth connection could not establish anymore. I hope that helps.

KM011092, 2024-01-28:
@gedongi : Have you been able to fix that already ?

gedongi, 2024-01-28:
Yes, i was able to fix it. Is depend on how you connect to the bluetti IOT device. I'am using ESP32 bridge. At this bridge there is a WiFi manager where you need to type the bluetti device name. I just replaced there the old with the new device name (bluetti bluetooth ID). The new name you will find inside bluetti app.
[image]

KM011092, 2024-01-28:
Ok, thanks. I am currently running the bluetti_mqtt code on a raspberry pi, that's a good opportunity to change that to ESP32 bridge, since I anyhow wanted to do that.

KM011092, 2024-01-30:
@gedongi Are you using the ESP32 bridge from mariolukas ? I have installed it and successfully connected it to my EP600, but can only see very basic EP600 information. Have you found a way to query the values I have added to this project using the ESP32 bridge ?