warhammerkid/bluetti_mqtt #67: Could not determine BlueZ version, bluetoothctl not available, assuming 5.51+

denisb88, 2023-03-06:
I get this error when I launch Bluetti to MQTT. How to fix it ?
Here is the log:
cont-init: info: /etc/cont-init.d/00-banner.sh exited 0
cont-init: info: running /etc/cont-init.d/01-log-level.sh
cont-init: info: /etc/cont-init.d/01-log-level.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[20:35:25] INFO: Reading configuration settings...
[20:35:28] INFO: Starting bluetti-mqtt...
2023-03-06 20:35:37 INFO     Connecting to MQTT broker...
2023-03-06 20:35:37 INFO     Starting to poll clients...
2023-03-06 20:35:37 INFO     Connecting to clients: ['04:7F:0E:A4:08:54']
2023-03-06 20:35:37 WARNING  Could not determine BlueZ version, bluetoothctl not available, assuming 5.51+
2023-03-06 20:35:37 INFO     Connected to MQTT broker
2023-03-06 20:35:37 INFO     Sent discovery message of AC200M-2227000041902 to Home Assistant
2023-03-06 20:35:37 WARNING  filtered_messages() is deprecated and will be removed in a future version. Use messages() together with Topic.matches() instead.
2023-03-06 20:35:38 ERROR    Error connecting to device 04:7F:0E:A4:08:54:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 40, in run
    await self.client.connect()
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 190, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local
2023-03-06 20:35:39 INFO     Retrying connection to 04:7F:0E:A4:08:54
2023-03-06 20:35:39 ERROR    Error connecting to device 04:7F:0E:A4:08:54:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 40, in run
    await self.client.connect()
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 190, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local

Alox-K, 2023-03-12:
Same here :-)
Would appreciate your help

warhammerkid, 2023-03-12:
What Linux distribution are you running? Did you install the BlueZ package?

Alox-K, 2023-03-12:
Thanks for quick response. Actually, I’m running your plug-in in home assistant, where it came integrated in SSMCD/bluetti2mqtt (https://github.com/SSMCD/bluetti2mqtt)
Do you have any idea to help me?

Alox-K, 2023-03-15:
OK, at least for my case it got solved after https://github.com/SSMCD/bluetti2mqtt got an update which included BlueZ package. Thank you so far.

closed 2023-05-13 by warhammerkid