warhammerkid/bluetti_mqtt #80: Device not found on Windows

atrueresistance, 2023-05-31:
[image]

Any ideas on why it might not show up? PC is same range as phone to the AC200MAX. Bluetooth is enabled.

bxm6306, 2023-06-01:
You can only connect one device to your power station at a time. If you recently used the Bluetti app on your phone, close it and disable your phone's bluetooth then try scanning again.

atrueresistance, 2023-06-01:
I did confirm that the AC200MAX Bluetooth state does say disconnected. The desktop is about 12 feet away from the Bluetti main unit, so perhaps it's too far away. I've an Orange Pi 3 LTS coming tomorrow to try in closer proximity. Will update once I receive it.

bxm6306, 2023-06-01:
On your PC, does your AC200Max show up as a Bluetooth device via the Add device wizard? (do not pair with it)
[image]

atrueresistance, 2023-06-01:
Nope, just one Unknown device. I'll try a laptop closer to the unit.
[image]

bxm6306, 2023-06-01:
Yeah, worth a try. Indoor range is heavily impacted by obstructions and other 2.4GHz signal interference as well as the sensitivity of your client's bluetooth transceiver.

closed 2023-06-01 by atrueresistance