warhammerkid/bluetti_mqtt #81: General Question About DATA TRANSFER

Aitomakethei, 2023-06-03:
In general: Is this project virtualizing the app-layer ?
Is it sending signals directly to the device in local network or does it sent, like app probably is ( ? ), to a middleman server from bluetti and only then back to the device in the home ?

Hope my questions make sense. 

Thank you

warhammerkid, 2023-06-04:
This project connects to your device directly using Bluetooth. The app supports connecting to your device over Bluetooth as well, but also supports connecting to a Bluetti server that the device also connects to over Wi-Fi.

Aitomakethei, 2023-06-04:
okay thank you. in that case it's interesting. would you say it's possible to create "custom made" Battery-Management-System with your interface? I'm thinking: feed or pull battery via socket based on information from energy meter.

warhammerkid, 2023-06-04:
Sure - a bunch of people have used Home Assistant as a platform for automations like this. Not every device has enough remote control to pull this off, though, so you may need an extra automated switch to disconnect it from the grid, for example. Also, not all the sensors Bluetti exposes are accurate enough to replace a true BMS (in the sense of estimating state of charge).

Aitomakethei, 2023-06-16:
my idea was to read the smartmeter IN/OUT of entire house and then dispense/charge the battery via a single socket based on that data from smartmeter. as long as your software here allows for remote control commands that dispense/charge via bluetooth based on this smartmeter data this should work? ( yes i'm assuming there's a little "loss" in energy - because battery isn't as close as possible to the solar generator in my scenario )

closed 2023-07-23 by warhammerkid