warhammerkid/bluetti_mqtt #82: The bluetti_mqtt does not work over the ESPHome bluetooth proxy

Karatzie, 2023-06-07:
Hello warhammerkid.

Thanks for your great Home Assistant integration for Bluetti powerstations. 
I am currently integrating the AC200Max into my Home Assistant system. 
Since I don't want to use the battery near the Home Assistant Raspberry Pi in different places, I have loaded the ESPHome Bluetooth Proxy on an ESP32. However, I can't get a connection to the Bluetti via this. Link: https://esphome.github.io/bluetooth-proxies/
When I load the Bluetooth Proxy's diagosis information from the Home Assistant's device list, I see that the Bluetooth Proxy has found the Bluetti. However, the bluetti_MQTT integration only seems to access the Bluetooth of the Raspberry Pi.
Is it possible to activate the Bluetooth Proxy in your bluetti_MQTT?

sibelle-labs, 2023-06-18:
Hi
the same problem and desire in my case ...
Great work !

bobthebuilder4711, 2023-07-19:
bluetti_mqtt is a standalone python program that sends its data to an MQTT Server of your choice. So there is no actual connection between bluetti_mqtt and Home Assistant.

What you're likely using is this wrapper, which really just makes it easier to install bluetti_mqtt on your home assistant instance, but doesn't really integrate it: https://github.com/SSMCD/bluetti2mqtt

You would have to ask the author of the wrapper integration you're using if they can find a way to present the ESPHome bluetooth proxy to bluetti_mqtt.

Patrick762, 2023-09-24:
Another option (in some places work in progress) is [this integration](https://github.com/Patrick762/hassio-bluetti-bt) I created recently, based on this repository