warhammerkid/bluetti_mqtt #84: Support for AC180

tonyk2023, 2023-06-18:
Can support for the new Bluetti AC180 please be added ?   Getting attached error message when trying to connect (I hope it's not a PEBKAC error)
[BluettiHomeAssistant.txt](https://github.com/warhammerkid/bluetti_mqtt/files/11780973/BluettiHomeAssistant.txt)

voltagex, 2023-06-24:
@tonyk2023 I don't have mine fully set up, but can you set up bluetti_mqtt (see the instructions in this repo) and run `bluetti-discovery  --log log-file.log E9:10:3D:EF:EC:44`

Then attach that log file to this issue.

I can supply a log, but only with a DC load attached.

tonyk2023, 2023-06-29:
Set up a fresh install using docker + HA so I could run the command with the following output.
[BluettiHomeAssistant2.txt](https://github.com/warhammerkid/bluetti_mqtt/files/11906728/BluettiHomeAssistant2.txt)

tonyk2023, 2023-06-29:
@voltagex Set up a fresh install using debian, docker + HA so I could run the command with the following output.

[BluettiHomeAssistant2.txt](https://github.com/warhammerkid/bluetti_mqtt/files/11910435/BluettiHomeAssistant2.txt)
[BluettiHomeAssistant3.txt](https://github.com/warhammerkid/bluetti_mqtt/files/11910437/BluettiHomeAssistant3.txt)

Pr3cis1on, 2023-07-19:
I've started a fork and have started working on AC180 support. I am very much "learning as I go", but I have my instance talking with my AC180. I'm planning on using the logger soon to try to get the AC180 file updated. I will post here with more soon.

voltagex, 2023-09-08:
@Pr3cis1on any updates?

voltagex, 2023-09-08:
@Pr3cis1on I'm not sure if it's your additional code or a problem in the original but in 4 minutes the script had sent almost 10 000 MQTT messages.

lenny1972, 2023-09-15:
> @Pr3cis1on any updates?

I'm also interested in having my new brand AC180 power station with my HA installation. In HA bluetti2mqtt configuration tab I've set the AC180 Bluetooth MAC Address and I can effectively connect with the AC180 but after the connection the integration crash and the Bluetti2mqtt service stop. When I come back home I post the log with a description of the error

tonyk2023, 2023-09-30:
So looking at the discovery output, how do you convert this into usable structures/data ?

{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAAAABhAo="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAAQAB1co="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAAgABJco="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAAwABdAo="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMABAABxcs="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMABQABlAs="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMABgABZAs="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMABwABNcs="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMACAABBcg="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMACQABVAg="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMACgABpAg="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMACwAB9cg="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMADAABRAk="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMADQABFck="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMADgAB5ck="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMADwABtAk="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCB9I6KQ==", "command": "AQMAEAABhc8="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAEQAB1A8="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAEgABJA8="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAEwABdc8="}
{"type": "client", "time": "2023-06-30 00:08:33 +1000", "data": "AQMCAAC4RA==", "command": "AQMAFAABxA4="}

I assume you can ignore the MODBUS Exception ReadHoldingRegisters ?

andersonas25, 2023-10-02:
I am using Ubuntu 22.04 so your directory my be different.

I have a ac180.py file that will stop the exceptions and provide a serial number and battery percentage.  (its a modified ep600.py file).  Rename the file ac180.py and place it in the /usr/local/lib/python3.10/dist-packages/bluetti_mqtt/core/devices directory

Also it will send a homeassistant discovery, and again only provide the serial number and battery percentage.

You also need to modify these file so that it will load the ac180.py file.
 pwd
/usr/local/lib/python3.10/dist-packages/bluetti_mqtt
grep -i -R ac180
bluetooth/__init__.py:from bluetti_mqtt.core import BluettiDevice, AC180, AC200M, AC300, AC500, AC60, EP500, EP500P, EP600, EB3A
bluetooth/__init__.py:DEVICE_NAME_RE = re.compile(r'^(AC180|AC200M|AC300|AC500|AC60|EP500P|EP500|EP600|EB3A)(\d+)$')
bluetooth/__init__.py:    if match[1] == 'AC180':
bluetooth/__init__.py:        return AC180(address, match[2])

So /bluetti-logger will give something like the following.
Waiting for connection...
Waiting for connection...
{'total_battery_percent': 65, 'device_type': 'AC180', 'serial_number': 23280000XXXX, 'power_generation': Decimal('0')}
{}
{}
{'device_type': 'AC180', 'serial_number': 23280000XXXXX}
{'power_generation': Decimal('0')}
{}
{}
{}
{'battery_range_start': 0, 'battery_range_end': 0}
{'max_ac_input_power': 0, 'max_ac_input_current': 0, 'max_ac_output_power': 0, 'max_ac_output_current': 0}
{}
{}
{}

As for other fields the only idea I have is to change the holding register and see if you get valid data.  Just trial and error I guess.

CCurvin, 2024-01-08:
Just chiming in to see if there's anything I can do to help move this forward and get it into a working release?

andersonas25, 2024-01-08:
I am able to use a modified ep600.py file as a ac180.py and got as far as the device type, serial number, and battery percent being sent home assistant.  Just trial and error I tried each .py file one at a time until I saw a discovery being sent.

From there I can take the output of bluetti-logger decode it from base64 but I don't know how to pull out the other fields.   So any hints would be appreciated.

bliekp, 2024-01-31:
Interesting, I also have an AC180 and HA. So far couldn't get this to work together. 
For bluetti_mqtt I have the issue that I have no idea where I find the device's MAC Address. I use the iPhone Bluetti app but it does not show this address anywhere...

voltagex, 2024-02-09:
@bliekp scan from your HA device

bliekp, 2024-02-09:
I did. Nothing.

bliekp, 2024-02-09:
Does not work. Also, tried it on a seperate Raspberry Pi and this also didn't find the device. Do I need to completely remove the device from the Phone app?

voltagex, 2024-02-09:
I don't have access to mine at the moment, I don't remember but I know the app could not be open while I scanned.

Will check again in a week.

tripzero, 2024-03-10:
> I am able to use a modified ep600.py file as a ac180.py and got as far as the device type, serial number, and battery percent being sent home assistant. Just trial and error I tried each .py file [...]

Even those fields would be a good start.  You should at least submit a pull request with what's working so far?

andersonas25, 2024-03-10:
I have never done a pull request before let me read up on the process.

alexnathanson, 2024-04-04:
I uncovered a few more registers for the AC180 (ac and dc power input and output) and made a pull request as well. The main data I need from the device now is the control registers for turning on and off the outputs. I think the easiest way to do that would be to find a tool like wireshark that can be run on a phone to grab the packets as you use the app, but I haven't found anything suitable. Any of y'all have any suggestions for a wireshark equivalent that runs on an android phone?

tripzero, 2024-04-06:
> I uncovered a few more registers for the AC180 (ac and dc power input and output) and made a pull request as well. The main data I need from the device now is the control registers for turning on [...]

Not sure completely, but maybe the "Enable Bluetooth HCI snoop log" option on android's Developer options?  Grok says you can then pull the log and open with wireshark with `adb pull /sdcard/btsnoop_hci.log`.  The answer is from grok so proceed with scrutiny.

bres55, 2024-04-25:
I used nRF Connect, on my android phone. It's fairly easy to use. You scan, connect, and log. Copy and then I emailed it to my self was the easiest way to get it on my PC.
It gave me a log, here's part of it, the bottom bit is when I started to use the Bluetti app

Haven't had time to look at it properly, but could be useful.
I'm, at the moment only keen to get the date from/to my AC180 via/to Arduino. The MQTT stuff is all a bit beyond me. I just want to switch a couple of relays on and off, under certain conditions.

Device Information (0x180A)
- System ID [R] (0x2A23)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Firmware Revision String [R] (0x2A26)
- Hardware Revision String [R] (0x2A27)
- Software Revision String [R] (0x2A28)
- Manufacturer Name String [R] (0x2A29)
Unknown Service (0000ff00-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000ff01-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [W WNR] (0000ff02-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [WNR] (0000ff03-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N W WNR] (0000ff04-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
Unknown Service (1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0)
- Unknown Characteristic [R W WNR] (f7bf3564-fb6d-4e53-88a4-5e37e0326063)
- Unknown Characteristic [R W WNR] (984227f3-34fc-4045-a5d0-2c581f81a153)
D	15:52:01.677	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	15:52:01.678	gatt.setCharacteristicNotification(0000ff01-0000-1000-8000-00805f9b34fb, true)
D	15:52:01.679	gatt.setCharacteristicNotification(0000ff04-0000-1000-8000-00805f9b34fb, true)
D	15:52:01.680	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	15:52:01.682	gatt.setCharacteristicNotification(0000ff01-0000-1000-8000-00805f9b34fb, true)
D	15:52:01.683	gatt.setCharacteristicNotification(0000ff04-0000-1000-8000-00805f9b34fb, true)
I	15:52:03.384	Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 01-03-7C-0D-45-00-3B-00-53-00-01-00-39-00-39-00-00-00-00-00-00-00-00-43-41-38-31-00-30-00-00-00-00-00-00-07-8B-5B-AF-02-20-00-00-00-01-00-01-00-01-00-49-00-23-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-6A-00-00-01-4D-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-01-00-01-A1-BE
A	15:52:03.384	"(0x) 01-03-7C-0D-45-00-3B-00-53-00-01-00-39-00-39-00-00-00-00-00-00-00-00-43-41-38-31-00-30-00-00-00-00-00-00-07-8B-5B-AF-02-20-00-00-00-01-00-01-00-01-00-49-00-23-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-6A-00-00-01-4D-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-01-00-01-A1-BE" received
I	15:52:03.431	Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 01-03-88-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-01-00-00-00-00-00-00-00-04-00-05-00-00-00-04-00-0F-00-01-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-05-89-05
A	15:52:03.431	"(0x) 01-03-88-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-01-00-00-00-00-00-00-00-04-00-05-00-00-00-04-00-0F-00-01-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-05-89-05" received
I	15:52:05.370	Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 01-03-7C-0D-45-00-3A-00-53-00-01-00-39-00-39-00-00-00-00-00-00-00-00-43-41-38-31-00-30-00-00-00-00-00-00-07-8B-5B-AF-02-20-00-00-00-01-00-01-00-01-00-49-00-23-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-6A-00-00-01-4D-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-01-00-01-6C-2E

bres55, 2024-04-30:
There may be some value in noting
 #Power IO
        self.struct.add_uint_field('output_mode',123) #32 when both loads off, 40 when AC is on, 48 when DC is on, 56 when both on 

off |    32 | 100000 | 1000
ac on | 40 | 101000 | 1010
dc on | 48 | 110000 | 1100
both | 56 | 111000 | 1110
in binary the high nibble look like flags

</body>

</html>

drjjr2, 2024-06-02:
Just got an AC180 and I can't get any information from the device at all. I get the following error:
```
2024-06-01 23:53:44 INFO     Connecting to MQTT broker...
2024-06-01 23:53:44 INFO     Starting to poll clients...
2024-06-01 23:53:44 INFO     Connecting to clients: ['D1:FD:5B:38:AB:12']
2024-06-01 23:53:44 INFO     Connected to MQTT broker
2024-06-01 23:53:51 INFO     Connected to device: D1:FD:5B:38:AB:12
2024-06-01 23:53:51 INFO     Device D1:FD:5B:38:AB:12 has name: AC1802411001681767
2024-06-01 23:53:51 WARNING  Delayed reconnect to D1:FD:5B:38:AB:12 after error
2024-06-01 23:53:57 INFO     Connected to device: D1:FD:5B:38:AB:12
2024-06-01 23:54:03 WARNING  Delayed reconnect to D1:FD:5B:38:AB:12 after error
2024-06-01 23:54:08 ERROR    Error connecting to device D1:FD:5B:38:AB:12:
Traceback (most recent call last):
  File "/home/don/.local/lib/python3.8/site-packages/bluetti_mqtt/bluetooth/client.py", line 78, in _connect
    await self.client.connect()
  File "/home/don/.local/lib/python3.8/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
  File "/home/don/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 254, in connect
    assert_reply(reply)
  File "/home/don/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort
```
AC180 shows
IoT version v9046.01
ARM version v2107.02
DSP version v2102.02
BMS version v1052.02

Anything that might point me in the right direction?

andersonas25, 2024-06-02:
Verify you have all the changes for the AC180 suppport.

Do you have anything else such as an app on your phone with a bluetooth connection to the AC180.   The AC180 can only handle one connection at a time.  How close is the mqtt-broker to the AC180.  Bluetooth is not long range and can be blocked by walls depending on what they are made of.

I have been using @alexnathanson latest code changes for a while now they work.

drjjr2, 2024-06-02:
> Verify you have all the changes for the AC180 suppport.

I am using the fork at [https://github.com/ftrueck/bluetti_mqtt/](https://github.com/ftrueck/bluetti_mqtt/). And I will check to make sure nothing else bluetooth is connected to it but I don't think so.

drjjr2, 2024-06-02:
> Do you have anything else such as an app on your phone with a bluetooth connection to the AC180. The AC180 can only handle one connection at a time. How close is the mqtt-broker to the AC180. Bluetooth is not long range and can be blocked by walls depending on what they are made of.
> 
Turned off Bluetooth on my phone. Same result. I also get the same result trying to connect via Bluetooth proxy and Home Assistant using the Bluetti-BT add-on. So two different devices with two different Bluetooth connections (not at the same time)
Could there be something different in the firmware for my AC180? Is there a log or something I can pull to help diagnose this?

andersonas25, 2024-06-02:
I did a complete fresh install on a spare PI4 of https://github.com/ftrueck/bluetti_mqtt instead of the modified 0.15 version I was running just to make sure it was OK.  It connects to my AC180 and mqtt-broker no problems.  

I believe the Bluetti-BT HACS integration is still bluetti_mqtt v0.15 so no AC180 support.  But I could be wrong.  

My AC180 has 
ARM V2073.04
DSP v2077.06
BSM v1033.12

According to the Bluetti android app the firmware is up to date.

Jun 02 15:17:14 raspberrypi4 bluetti-mqtt[23816]: 2024-06-02 15:17:14 INFO     Connected to device: F6:9C:08:XX:XX:XX
Jun 02 15:17:14 raspberrypi4 bluetti-mqtt[23816]: 2024-06-02 15:17:14 INFO     Device F6:9C:08:XX:XX:XX has name: AC18023280000XXXXX
Jun 02 15:17:14 raspberrypi4 bluetti-mqtt[23816]: 2024-06-02 15:17:14 INFO     Sent discovery message of AC180-23280000XXXXXto Home Assistant

Not sure what to say, unless there is something blocking bluetooth.

May if you share more info on what you are running the bluetti_mqtt code on.

drjjr2, 2024-06-02:
> May if you share more info on what you are running the bluetti_mqtt code on.

It's a Gigabyte GB-BXi3H-5010 with the onboard Bluetooth 4.0. So...is that the problem? It's pretty old.
```
Agent registered
[CHG] Controller F4:06:69:17:5B:50 Pairable: yes
[bluetooth]# show
Controller F4:06:69:17:5B:50 (public)
	Name: Torrent
	Alias: Torrent
	Class: 0x000c0104
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0535
	Discovering: no
Advertising Features:
	ActiveInstances: 0x00
	SupportedInstances: 0x05
	SupportedIncludes: appearance
	SupportedIncludes: local-name
```

drjjr2, 2024-06-03:
Back to trying the Bluetooth proxy, I see this in the log. Anything helpful?
```
20:57:24][I][bluetooth_proxy:282]: [0] [D1:FD:5B:38:AB:12] Connecting v3 without cache
[20:57:24][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[20:57:24][I][esp32_ble_client:067]: [0] [D1:FD:5B:38:AB:12] 0x00 Attempting BLE connection
[20:57:24][D][esp32_ble_client:110]: [0] [D1:FD:5B:38:AB:12] ESP_GATTC_CONNECT_EVT
[20:57:24][D][esp32_ble_client:110]: [0] [D1:FD:5B:38:AB:12] ESP_GATTC_OPEN_EVT
[20:57:24][D][esp32_ble_tracker:266]: Starting scan...
[20:57:24][D][esp32_ble_client:110]: [0] [D1:FD:5B:38:AB:12] ESP_GATTC_SEARCH_CMPL_EVT
[20:57:24][I][esp32_ble_client:227]: [0] [D1:FD:5B:38:AB:12] Connected
[20:57:24][D][esp32_ble_client:188]: [0] [D1:FD:5B:38:AB:12] cfg_mtu status 0, mtu 517
[20:57:24][D][esp32_ble_client:110]: [0] [D1:FD:5B:38:AB:12] ESP_GATTC_REG_FOR_NOTIFY_EVT
[20:57:24][D][esp32_ble_client:110]: [0] [D1:FD:5B:38:AB:12] ESP_GATTC_WRITE_DESCR_EVT
[20:57:24][D][esp-idf:000][BTU_TASK]: W (42664) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x13

[20:57:24][D][esp-idf:000][BTU_TASK]: W (42669) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13

[20:57:24][D][esp-idf:000][BTU_TASK]: W (42670) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x13

[20:57:24][D][esp-idf:000][BTU_TASK]: W (42672) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13

[20:57:24][D][esp32_ble_client:110]: [0] [D1:FD:5B:38:AB:12] ESP_GATTC_WRITE_CHAR_EVT
[20:57:24][W][bluetooth_proxy.connection:081]: [0] [D1:FD:5B:38:AB:12] Error writing char/descriptor at handle 0x22, status=133
[20:57:24][D][esp32_ble_client:172]: [0] [D1:FD:5B:38:AB:12] ESP_GATTC_DISCONNECT_EVT, reason 19
[20:57:24][D][esp32_ble_client:110]: [0] [] ESP_GATTC_CLOSE_EVT
```

andersonas25, 2024-06-03:
What version of the OS and python on the gigabyte running bluetti-mqtt?

So the Gigabyte running the bluetti-mqtt python code and the ESP32 both cannot connect to the AC180?  And you only have one running at a time?  Or are you expecting the esp32 to somehow proxy the connection for the gigabyte?

drjjr2, 2024-06-03:
> So the Gigabyte running the bluetti-mqtt python code and the ESP32 both cannot connect to the AC180? And you only have one running at a time? Or are you expecting the esp32 to somehow proxy the connection for the gigabyte?

The Gigabyte won't connect to it by itself. Actually tried two different Gigabyte units (I just happen to have them unused and they are running Ubuntu). The esp32 bluetooth proxy is separate and I don't have them both trying to connect at the same time. When I couldn't get bluetti-mqtt to connect I tried the HA/proxy route. 

I have an Android tablet and nRF Connect will connect to the AC180 but there ends my knowledge of Bluetooth and what information is significant.

jsaiko, 2024-06-14:
@drjjr2 any luck?  I am having the same issue using the fork at https://github.com/ftrueck/bluetti_mqtt/

drjjr2, 2024-06-15:
> @drjjr2 any luck? I am having the same issue using the fork at https://github.com/ftrueck/bluetti_mqtt/

No luck. I suspect there's something that's changed with the firmware but I have nothing to back that up. 3 different devices with different Bluetooth hardware (including esp32 Bluetooth proxy) have been unable to connect using bluetti_mqtt but the bluetti app works fine.

jsaiko, 2024-06-15:
> No luck. I suspect there's something that's changed with the firmware but I have nothing to back that up. 3 different devices with different Bluetooth hardware (including esp32 Bluetooth proxy) have been unable to connect using bluetti_mqtt but the bluetti app works fine.

Same.  I ordered an RF dongle and I'm going to try to sniff the packets to see what's going on.