warhammerkid/bluetti_mqtt #86: Software caused connection abort

iandees, 2023-07-26:
I'm running bluetti-mqtt with:

```
bluetti-mqtt --broker 127.0.0.1 70:B8:F6:80:97:xx --interval 30
```

and get the following error every second or two:

```
Traceback (most recent call last):
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bluetti_mqtt/bluetooth/client.py", line 78, in _connect
    await self.client.connect()
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 190, in connect
    assert_reply(reply)
  File "/home/iandees/bluetti-venv/lib/python3.8/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort
```

iandees, 2023-07-26:
It seems that this might have been solved by cycling bluetooth on/off on the AC500 it's connecting to.

ceredoandy, 2024-03-14:
I'm having the same problem cycling blutooth did not help