warhammerkid/bluetti_mqtt #89: EB3A - Connection issue

yugrinkov, 2023-09-26:
Hi guys,

I have encountered with next issue using this package on my workstation.

```
bluetti-mqtt --scan                                                       
We renamed asyncio-mqtt to aiomqtt and released a version 1.0.0 in the process. This is the last release under the asyncio-mqtt name. You can find the new repository at https://github.com/sbtinstruments/aiomqtt
Scanning....
Found EB3A2236001325010: address CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0
```
The address looks like invalid. Do you have any ideas what went wrong?

macOS Ventura Version 13.4.1 (c) (22F770820d)

TylerBizz, 2023-09-28:
I have a similar type of address. which appears to the the UUID. I'm not able to connect.

TylerBizz, 2023-09-28:
> You are able to connect as evident by this line:
> 
> > 2023-09-28 17:05:17 INFO Connected to device: B800XXXX-0675-A28E-1781-XXXXXXXXXXXX
> 
> After the Bluetooth client connects, it asks the device for its name by using "00002a00-0000-1000-8000-00805f9b34fb". And that's where the error is produced:
> 
> > bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
> 
> I've got no macOS, so I wouldn't be able to reproduce this error even if I knew what I was doing.
> 
> What you can try is to create a Bluetooth log as explained in the readme.rst so maybe others can help figure out what needs to be done.

Thanks and agreed - I missed that it was connecting. I managed to get a virtual machine with Windows setup and installed Bluetti-mqtt. The "bluetti-mqtt --scan" process produces different results when run on Mac and Windows. ie. The device name and address that it returns is different. So, I was able to get the address that looked more like others in videos and documentation however the same issue is experienced. 

Looking at the error in terminal and watching the display on the AC500 I also noticed that it was connecting and then dropping a sec later and retrying but failing to retrieve the device name. I'll have a look at creating a bluetooth log.

bobthebuilder4711, 2023-09-28:
From what I've found online, macOS doesn't give out the MAC addresses of Bluetooth devices. Instead it gives you UUIDs. But the macOS version of the bleak library knows this and simply works with UUIDs instead of MAC addresses. So finding the MAC in Windows and then using that on macOS is likely going to fail.

The Bluetooth log might have the answer. Start the logging and then open bluetti's app and connect to your AC500 via Bluetooth. I haven't used the app in a long time, but I think it asks for the device's name, so there should be something about that in the log.

TylerBizz, 2023-09-28:
Yeah, I've come to the same conclusion. I initially thought Bluetooth was more straight forward but its been an educational 24 hours :) I'll try the logging over the weekend.

yugrinkov, 2023-09-29:
Hi guys, 

Unfortunately reverse engineering logs can't be created for me :( I got the same error, see below: 

```
bluetti-discovery --log the-log-file.log CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0
Connecting to CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0
Waiting for connection...
ERROR:root:Error retrieving device name CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 90, in _get_name
    name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/__init__.py", line 711, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
WARNING:root:Delayed reconnect to CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0 after error
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
ERROR:root:Error retrieving device name CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 90, in _get_name
    name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/__init__.py", line 711, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
```

TylerBizz, 2023-09-29:
Hmmm … unable to connect for logging on Mac for my AC500. I also tried through virtual machine running Windows 11, scanned and then tried logging but also failing to retrieve device name. When trying on Windows VM, the AC500 briefly shows connect on the lcd but then disconnects with same error - failing to retrieve device name but then VM crashes.  Will post the error shortly.

TylerBizz, 2023-09-29:
bluetti-mqtt --scan

We renamed asyncio-mqtt to aiomqtt and released a version 1.0.0 in the process. This is the last release under the asyncio-mqtt name. You can find the new repository at https://github.com/sbtinstruments/aiomqtt
Scanning....
Found AC500XXXXXXXXXXXXX: address B8001FB6-0675-A28E-1781-DD57522D886F

bluetti-logger --log the-log-file.log B8001FB6-0675-A28E-1781-DD57522D886F

Connecting to B8001FB6-0675-A28E-1781-DD57522D886F
Waiting for connection...
Waiting for connection...
ERROR:root:Error retrieving device name B8001FB6-0675-A28E-1781-DD57522D886F:
Traceback (most recent call last):
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 90, in _get_name
    name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/__init__.py", line 711, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
WARNING:root:Delayed reconnect to B8001FB6-0675-A28E-1781-DD57522D886F after error
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
ERROR:root:Error retrieving device name B8001FB6-0675-A28E-1781-DD57522D886F:
Traceback (most recent call last):
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 90, in _get_name
    name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/__init__.py", line 711, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
WARNING:root:Delayed reconnect to B8001FB6-0675-A28E-1781-DD57522D886F after error
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
ERROR:root:Error retrieving device name B8001FB6-0675-A28E-1781-DD57522D886F:
Traceback (most recent call last):
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 90, in _get_name
    name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/__init__.py", line 711, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
WARNING:root:Delayed reconnect to B8001FB6-0675-A28E-1781-DD57522D886F after error
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
ERROR:root:Error retrieving device name B8001FB6-0675-A28E-1781-DD57522D886F:
Traceback (most recent call last):
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py", line 90, in _get_name
    name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/__init__.py", line 711, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
WARNING:root:Delayed reconnect to B8001FB6-0675-A28E-1781-DD57522D886F after error
Waiting for connection...
Waiting for connection...
[... 103 lines cut ...]
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/__init__.py", line 711, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
WARNING:root:Delayed reconnect to B8001FB6-0675-A28E-1781-DD57522D886F after error
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
Waiting for connection...
ERROR:root:Error retrieving device name B8001FB6-0675-A28E-1781-DD57522D886F:
Traceback (most recent call last):
[... cut ...]
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/Users/vec/.pyenv/versions/3.10.6/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 276, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb was not found!
WARNING:root:Delayed reconnect to B8001FB6-0675-A28E-1781-DD57522D886F after error
Waiting for connection...
Waiting for connection...
Waiting for connection...

TylerBizz, 2023-10-03:
Any solution on running Bluetti_MQTT on Mac Silicon??

yugrinkov, 2023-10-03:
> Any solution on running Bluetti_MQTT on Mac Silicon?? I'm still unable to connect - even just to run the logger.

The same for me. I have come to solution is buy raspberry pi and run it there.

TylerBizz, 2023-10-04:
@yugrinkov I got mine to work last night. 

The Bluetti_MQTT code is looking for the UUID 00002a00-0000-1000-8000-00805f9b34fb which is assuming it is standard across Bluetti devices. This is supposed to be a published bluetooth "characteristic" that the code then decodes to the device name. This is not happening per the error message. Not sure if it is a Mac thing. 

The device address you are getting is not wrong and looks different to windows machines but it is a useable address on Mac. 

What I did to get around the issue was locate the file called "client.py" in Bluetti-mqtt/bluetooth/ folder. You can find this by locating where your python library you are using is saved. Depends on how you have your environment setup. 

Open that file in Xcode or text editor - you will see the reference to "DEVICE_NAME_UUID = '00002a00-0000-1000-8000-00805f9b34fb'" on line 25. I left that as is and edited line 90 & 91 from:

name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
self.name = name.decode('ascii')
    
to
       
name = 'AC5002123' 
self.name = 'AC5002123'

The 2123 is the first 4 digits on the serial number after the model number - this was needed as it didn't work with just AC500. The EB3A is listed in the code as one of the devices so for yours you should try  'EB3AXXXX' with the Xs being the first 4 digits after 'EB3A' the 'Bluetti-mqtt --scan' command returns to you as per your first comment. Save the file. If you want to retain the original lines then duplicate them and edit one version and comment out the other with a # in front of the line. So if something doesn't work you can always return the code to how it was.
eg. 

#name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID)
name = 'AC5002123' 
#self.name = name.decode('ascii')
self.name = 'AC5002123'

This bypasses the search and return of the device name and forced it to use my known device name. 

I tested this on the logger first, which for you should be:
bluetti-logger --log the-log-file.log CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0

If it works there then you should be able to go ahead with the mqtt service. 

If it doesn't work then try only 3 numbers after EB3A as I don't know how many numbers are used to reference the model for that device. I worked mine out through trial and error.

yugrinkov, 2023-10-04:
> @yugrinkov I got mine to work last night.
> 
> The Bluetti_MQTT code is looking for the UUID 00002a00-0000-1000-8000-00805f9b34fb which is assuming it is standard across Bluetti devices. This is supposed to be a published bluetooth "characteristic" that the code then decodes to the device name. This is not happening per the error message. Not sure if it is a Mac thing.
> 
> The device address you are getting is not wrong and looks different to windows machines but it is a useable address on Mac.
> 
> What I did to get around the issue was locate the file called "client.py" in Bluetti-mqtt/bluetooth/ folder. You can find this by locating where your python library you are using is saved. Depends on how you have your environment setup.
> 
> Open that file in Xcode or text editor - you will see the reference to "DEVICE_NAME_UUID = '00002a00-0000-1000-8000-00805f9b34fb'" on line 25. I left that as is and edited line 90 & 91 from:
> 
> name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID) self.name = name.decode('ascii')
> 
> to
> 
> name = 'AC5002312' self.name = 'AC5002312'
> 
> The 2312 is the first 4 digits on the serial number after the model number - this was needed as it didn't work with just AC500. The EB3A is listed in the code as one of the devices so for yours you should try 'EB3AXXXX' with the Xs being the first 4 digits after 'EB3A' the 'Bluetti-mqtt --scan' command returns to you as per your first comment. Save the file. If you want to retain the original lines then duplicate them and edit one version and comment out the other with a # in front of the line. So if something doesn't work you can always return the code to how it was. eg.
> 
> #name = await self.client.read_gatt_char(self.DEVICE_NAME_UUID) name = 'AC5002312' #self.name = name.decode('ascii') self.name = 'AC5002312'
> 
> This bypasses the search and return of the device name and forced it to use my known device name.
> 
> I tested this on the logger first, which for you should be: bluetti-logger --log the-log-file.log CD9BF1CC-4B49-7811-68E2-570B5A7D0CB0
> 
> If it works there then you should be able to go ahead with the mqtt service.
> 
> If it doesn't work then try only 3 numbers after EB3A as I don't know how many numbers are used to reference the model for that device. I worked mine out through trial and error.

Thank you! Amazing job! It works for me now!

TylerBizz, 2023-10-06:
@yugrinkov Awesome! :) 

There still needs to be a correction in the code to find the UUID that responds with the Model Number on MacOS so I will leave this open to see if someone else can assist. I cannot get the characteristic required to show when scanning the device from a Mac for published bluetooth data. So maybe an adjustment that asks for the Model number based on the Bluetti machine type can be input on request when running the script if there is no other way to call it on a Mac. But this kind of fix is a little beyond my knowledge at this time. I have also reached out to Bluetti to see if there is a change in the bluetooth characteristics published and to see if they can provide any other data.

LojzePek, 2025-02-02:
I see this post is a bit old. But thanks to **TylerBizz** i finally managed to connect AC180 with my Mac :). 
But after executing bluetti-mqtt i get stuck here:
`2025-02-02 12:03:10 INFO     Connecting to MQTT broker...
2025-02-02 12:03:10 INFO     Starting to poll clients...
2025-02-02 12:03:10 INFO     Connecting to clients: ['0B552FEB-F424-0860-75AA-0901F11CB767']
2025-02-02 12:03:10 INFO     Connected to MQTT broker
2025-02-02 12:03:16 INFO     Connected to device: 0B552FEB-F424-0860-75AA-0901F11CB767
2025-02-02 12:03:16 INFO     Device 0B552FEB-F424-0860-75AA-0901F11CB767 has name: AC1802427
2025-02-02 12:03:17 WARNING  Delayed reconnect to 0B552FEB-F424-0860-75AA-0901F11CB767 after error
2025-02-02 12:03:23 INFO     Connected to device: 0B552FEB-F424-0860-75AA-0901F11CB767`

Nothing is send to my MQTT server....

TylerBizz, 2025-02-03:
> I see this post is a bit old. But thanks to **TylerBizz** i finally managed to connect AC180 with my Mac :). But after executing bluetti-mqtt i get stuck here:
> 
> 2025-02-02 12:03:10 INFO Connecting to MQTT broker... 2025-02-02 12:03:10 INFO Starting to poll clients... 2025-02-02 12:03:10 INFO Connecting to clients: ['0B552FEB-F424-0860-75AA-0901F11CB767'] 2025-02-02 12:03:10 INFO Connected to MQTT broker 2025-02-02 12:03:16 INFO Connected to device: 0B552FEB-F424-0860-75AA-0901F11CB767 2025-02-02 12:03:16 INFO Device 0B552FEB-F424-0860-75AA-0901F11CB767 has name: AC1802427 2025-02-02 12:03:17 WARNING Delayed reconnect to 0B552FEB-F424-0860-75AA-0901F11CB767 after error 2025-02-02 12:03:23 INFO Connected to device: 0B552FEB-F424-0860-75AA-0901F11CB767
> 
> This is logger output: Connecting to 0B552FEB-F424-0860-75AA-0901F11CB767 Waiting for connection... Waiting for connection... WARNING:root:Delayed reconnect to 0B552FEB-F424-0860-75AA-0901F11CB767 after error Got an error running command ReadHoldingRegisters(starting_address=0, quantity=21): disconnected
> 
> Nothing is send to my MQTT server....

It's been a long time since I used this so I will have to try reconnect on mine and see if everything is still working. Not sure what is occurring according to your error. 

what is your bluetti-mqtt excution command, with all your arguments? (removing any password/user details)

does scan and logger work for you?

LojzePek, 2025-02-05:
Thanks Tyler for the help. I have my mac 10 cm from buletti, so distance should not be an issue ;). 
So bluetti-discovery --scan and bluetti-discovery --scan find my bluetti with the same address. This part is ok.

And bluetti-mqtt --broker .... call should be correct since the script is connecting to my mqtt server (if i change the password to incorrect value script outputs MQTT error  ). So this part i guess is also valid. And the bluetooth icon is activated on bluetti after running the call.

I tried again both calls. Both actives the BT icon on my bluetti

bluetti-mqtt --broker mqtt.xxx.com --username xxx --password xxx -- 60 0B552FEB-F424-0860-75AA-0901F11CB767
2025-02-05 09:41:02 INFO     Connecting to MQTT broker...
2025-02-05 09:41:02 INFO     Starting to poll clients...
2025-02-05 09:41:02 INFO     Connecting to clients: ['0B552FEB-F424-0860-75AA-0901F11CB767', '60']
2025-02-05 09:41:02 INFO     Connected to MQTT broker
2025-02-05 09:41:08 INFO     Connected to device: 0B552FEB-F424-0860-75AA-0901F11CB767
2025-02-05 09:41:08 INFO     Device 0B552FEB-F424-0860-75AA-0901F11CB767 has name: AC1802427
2025-02-05 09:41:09 WARNING  Delayed reconnect to 0B552FEB-F424-0860-75AA-0901F11CB767 after error
2025-02-05 09:41:15 INFO     Connected to device: 0B552FEB-F424-0860-75AA-0901F11CB767
Here the script is stuck, but the bluetooth icon on bluetti is still active. I must disable BT on mac to get connection released

bluetti-logger --log log-file.log 0B552FEB-F424-0860-75AA-0901F11CB767
Connecting to 0B552FEB-F424-0860-75AA-0901F11CB767
Waiting for connection...
Waiting for connection...
WARNING:root:Delayed reconnect to 0B552FEB-F424-0860-75AA-0901F11CB767 after error
Got an error running command ReadHoldingRegisters(starting_address=0, quantity=21): disconnected
The log file is sadly empty 

I tried also on windows 11 and ubuntu that runs on parallels but, the were other issues with bluetooth so i have given up.

There is also a git project for esp32 (https://github.com/mariolukas/Bluetti_ESP32_Bridge ) that runs on arduino but sadly this project is not updated and uses some deprecated libraries. So i could not compile it in Arduino :)

If i would be 100% sure that i could resolve the issue with purchasing raspberry PI and running the script there, i would go this route, but for now i don't have the energy to invest the time on this :).

Thanks anyway.

TylerBizz, 2025-02-06:
The AC180 is not supported and you need to add it using the bluetti-discovery —scan option to get the device address, followed by adding the address and log argument to the same command (refer to my post above) Have AC and DC loads on the device. The program will then scan your device and map it accordingly. It looks like you only did the first step to get your address. A raspberry pi or alternate operating system will not solve it as your device needs to be a recognized device in the code base to be able to communicate with mqtt. The Bluetooth connection is working this is a good sign and means the your device can be read but follow the steps I learned in my previous post and you should be able to map and add support for the AC180. (At least the instructions claim it to be so)