warhammerkid/bluetti_mqtt #90: Connecting to AC200M

Paul-9929, 2023-10-27:
I running the python on a windows machine and can scan and fine my Bluetti device. When running the broker command it connect to my MQTT broker but then crashes. On my broker I see the connection, it is accepted by the broker, a subscription is created. Then the client disconnects. 
Any assistance would be greatly appreciated. 

Command trace...
C:\Windows\System32>bluetti-mqtt --broker 192.168.1.12 --username Bluetti --password Bluetti --interval 10 04:7F:0E:A3:E8:E4
We renamed asyncio-mqtt to aiomqtt and released a version 1.0.0 in the process. This is the last release under the asyncio-mqtt name. You can find the new repository at https://github.com/sbtinstruments/aiomqtt
2023-10-27 15:54:53 INFO     Connecting to MQTT broker...
2023-10-27 15:54:53 INFO     Starting to poll clients...
2023-10-27 15:54:53 INFO     Connecting to clients: ['04:7F:0E:A3:E8:E4']
2023-10-27 15:54:53 INFO     Connected to MQTT broker
2023-10-27 15:54:53 WARNING  filtered_messages() is deprecated and will be removed in a future version. Use messages() together with Topic.matches() instead.
2023-10-27 15:55:00 INFO     Connected to device: 04:7F:0E:A3:E8:E4
2023-10-27 15:55:00 INFO     Device 04:7F:0E:A3:E8:E4 has name: AC200M2224002627157
2023-10-27 15:55:00 ERROR    Crashing with uncaught exception:
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\bluetti_mqtt\device_handler.py", line 31, in run
    await asyncio.gather(*(polling_tasks + pack_polling_tasks + [manager_task]))
  File "C:\Program Files\Python312\Lib\site-packages\bluetti_mqtt\device_handler.py", line 65, in _pack_poll
    device = self._get_device(address)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\bluetti_mqtt\device_handler.py", line 103, in _get_device
    self.devices[address] = build_device(address, name)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\bluetti_mqtt\bluetooth\__init__.py", line 28, in build_device
    if match[1] == 'AC200M':
       ~~~~~^^^
TypeError: 'NoneType' object is not subscriptable
2023-10-27 15:55:00 INFO     Shutting down...

bobthebuilder4711, 2023-11-02:
What version are you using? The current repository doesn't have a "\bluetti_mqtt\bluetooth_init.py"

Instead the code is found in "bluetti_mqtt/bluetooth/__init__.py" now.

So simply updating might do the trick.

If you used "pip install bluetti-mqtt" then use the most recent release instead:
"pip install https://github.com/warhammerkid/bluetti_mqtt/archive/refs/tags/v0.15.0.zip"

Paul-9929, 2023-11-06:
Thanks for the response I was installing using pip install bluetti-mqtt and then as you indicated i installed the latest version from the github url. But I'm still getting the same error.

bobthebuilder4711, 2023-11-06:
Then it's probably the same problem as the macOS guys are having: Windows doesn't want to tell you the name of the device.