warhammerkid/bluetti_mqtt #98: What would be involved to add support for EP800?

jhagenk, 2024-01-26:
I have an EP800 and would love to have support for this in HomeAssistant. What would be involved in adding support?

After communication with Bluetti, I'm 99% sure that the physical inverter box between the EP600, EP760, EP800, and EP900 are all the same controllers, just with different spec inverters for the different power standards followed by each machine. 

If the project were to take the existing EP600 integration and also enable that for EP760, EP800, EP900 then I bet it would work. 

I have an EP800 to test it with. I do not have an EP760 or EP900.

Patrick762, 2024-02-05:
I'm currently testing the EP760 with the fields from the EP600 on the Home Assistant Integration. Should be the same, because both use v2 of the bluetti field tables.
If you're using Home Assistant, I could add the EP800 to the next beta for you to test.

jhagenk, 2024-02-05:
I’m on the Home Assistant Yellow hardware. I’ll be happy to help test it out.

Thanks!
-Jeff

Patrick762, 2024-02-25:
@jhagenk There's now a beta release you can test in HACS:

https://github.com/Patrick762/hassio-bluetti-bt/releases/tag/0.0.18

This should add basic data about the EP800. If there are any problems, please create an issue in the integration repository or re-open the issue https://github.com/Patrick762/hassio-bluetti-bt/issues/59