warhammerkid/bluetti_mqtt #102: Errors when connecting (EP500Pro) on Raspberry Pi

swoop124, 2024-02-14:
Hello everyone,

i have a EP500Pro, til now i connected through my Linux Notebook to the Bluetti device, it works great, and all data are sent to HA.
But now i switched to a freshly installed Raspberry PI 3B (Rasbian aarch64) and get some errors.

First I installed python-pip:
`apt install python3-pip`
 with all dependencies. 

Then i installed bluetti-mqtt:
`pip install bluetti_mqtt`
it all went well

When I enter `bluetti-mqtt  --scan`
I get the right device-id.
Then i used this command to establish the connection to my mqtt broker/HA:
`bluetti-mqtt --broker 10.10.201.2 --username mqtt-user --password <password>  DA:9F:10:22:33:44`

but i get this errors:
```
We renamed asyncio-mqtt to aiomqtt and released a version 1.0.0 in the process. This is the last release under the asyncio-mqtt name. You can find the new repository at https://github.com/sbtinstruments/aiomqtt
2024-02-14 13:58:27 INFO     Connecting to MQTT broker...
Exception ignored in: <function Client.__del__ at 0x7fa7a97c10>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
2024-02-14 13:58:27 INFO     Starting to poll clients...
2024-02-14 13:58:27 ERROR    Crashing with uncaught exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/bluetti_mqtt/mqtt_client.py", line 513, in run
    async with Client(
  File "/usr/local/lib/python3.9/dist-packages/asyncio_mqtt/client.py", line 298, in __init__
    self._client: mqtt.Client = mqtt.Client(
TypeError: __init__() missing 1 required positional argument: 'callback_api_version'
2024-02-14 13:58:27 INFO     Connecting to clients: ['DA:9F:10:55:08:FC']
2024-02-14 13:58:27 INFO     Shutting down...
```
Is there something missing?
Any ideas to resolve this issue?

thanks

nicolasrata, 2024-02-15:
Same problem with AC200Max

Maybe a solution here
[(https://github.com/eclipse/paho.mqtt.python/releases/tag/v2.0.0)]

swoop124, 2024-02-16:
Hello and thank you for your answer.

But it seams this version is already installed.

```
# pip install paho-mqtt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.9/dist-packages (2.0.0)
```

so, any other ideas?

In the meantime, I tested RaspberryOS in 32bit, but no changes.

swoop124, 2024-02-20:
I was testing and installed a Intel NUC5, but i got the same error.
so i checked the bluetooth-versions. 
My Notebook has 4.2 with this, it's working
Raspberry Pi has 4.1 (is not working)
and the NUC has 4.0 (also not working)
could it be the version of bluetooth?

i'll try with a RB4, if i get one...

nicolasrata, 2024-02-20:
I'm using bluetti_mqtt on a RB3 with raspbian I think and it's working.
I tried to install dietPi on the same RB and it doesn't work.

I will check the version of the different packages

AR-General, 2024-02-20:
Hi, on my fresh installation paho-mqtt version 2.0.0 was installed automatically.
Downgrading it with `pip install paho-mqtt==1.6.1` solved the issue for now

swoop124, 2024-02-20:
hello,

i can confirm this.
my RPi3B is now working as expected.

thank you very much

closed 2024-02-20 by swoop124