warhammerkid/bluetti_mqtt #104: Suport for AC200L

Kathou99, 2024-02-25:
Hi, with the AC200L I can't connect with the HA addon, see below the output I got.

Run the the HA addon

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/00-banner.sh

-----------------------------------------------------------
 Add-on: Bluetti to MQTT
 MQTT bridge between Bluetti and Home Assistant
-----------------------------------------------------------
 Add-on version: 1.4.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.5  (amd64 / qemux86-64)
 Home Assistant Core: 2024.2.3
 Home Assistant Supervisor: 2024.02.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
cont-init: info: /etc/cont-init.d/00-banner.sh exited 0
cont-init: info: running /etc/cont-init.d/01-log-level.sh
cont-init: info: /etc/cont-init.d/01-log-level.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[14:10:13] INFO: Reading configuration settings...
[14:10:14] INFO: Starting bluetti-logger...
[14:10:14] INFO: Messages are NOT published to the MQTT broker in logger mode.
Traceback (most recent call last):
  File "/usr/local/bin/bluetti-logger", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/site-packages/bluetti_mqtt/logger_cli.py", line 119, in main
    asyncio.run(log(args.address, args.log))
  File "/usr/local/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.10/site-packages/bluetti_mqtt/logger_cli.py", line 55, in log
    devices = await check_addresses({address})
  File "/usr/local/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/__init__.py", line 55, in check_addresses
    return [build_device(d.address, d.name) for d in filtered]
  File "/usr/local/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/__init__.py", line 55, in <listcomp>
    return [build_device(d.address, d.name) for d in filtered]
  File "/usr/local/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/__init__.py", line 28, in build_device
    if match[1] == 'AC200M':
TypeError: 'NoneType' object is not subscriptable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

filippo5589, 2024-04-02:
Hi I also have the same problem with AC200L, you can't connect

filippo5589, 2024-04-02:
[image]

ftrueck, 2024-04-07:
There are lots of changes required. Basically the AC200L is similar to the AC200M. There are some minor quirks but basically it uses the same code. One of the bigger problems though is that the mqtt code seems in need to be fixed. In my local version I had lots of issues getting it to work.

MikeWolfeGHC, 2024-04-17:
@ftrueck were you able to get it to work?  I have the same problem, and any insight you can give would be appreciated.

tripzero, 2024-04-29:
There is this: https://github.com/warhammerkid/bluetti_mqtt/pull/106

I wonder if there are controls for ac input.

ftrueck, 2024-05-03:
@MikeWolfeGHC Here is my working fork with support for AC200L:

[https://github.com/ftrueck/bluetti_mqtt](https://github.com/ftrueck/bluetti_mqtt)

ftrueck, 2024-05-03:
@tripzero there are controls for:
- AC Output
- DC Output
- Power Off
- Display Timeout

The input can not be changed at the moment.

MikeWolfeGHC, 2024-05-03:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thank you

Mike Wolfe
Vice President - Technology
George Harms Construction Co., Inc.

https://github.com/ftrueck/bluetti_mqtt

filippo5589, 2024-05-03:
Ciao io non riesco a farlo funzionare riuscite ad aiutarmi? @ftrueck  non riesco ad inserire la tua repository.
[image]

tripzero, 2024-05-03:
> @tripzero there are controls for:
> 
> * AC Output
> * DC Output
> * Power Off
> * Display Timeout
> 
> The input can not be changed at the moment.

Is that an bluetooth API thing or just we lack the right commands?  I can't control it from the app, so I'm guessing it's the former...

ftrueck, 2024-05-04:
> Hi, I can't make it work, can you help me? @ftrueck I can't enter your repository. <img alt="image" width="1076" src="https://private-user-images.githubusercontent.com/120588148/327892989-eb015115-d98e-4c96-b482-c54eb64a5713.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ4MDg0NjMsIm5iZiI6MTcxNDgwODE2MywicGF0aCI6Ii8xMjA1ODgxNDgvMzI3ODkyOTg5LWViMDE1MTE1LWQ5OGUtNGM5Ni1iNDgyLWM1NGViNjRhNTcxMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwNFQwNzM2MDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNjYzNWM4OGM5ZjU5ODE3NzRkMjEwNDk1MzUxZWQ1MzQwYWExOTk3YjUxYmM2YzlhMDliZjI1MDIxNjM4YWJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MtK10ZTGAN2e3XvQOcc0GsZ8AGzrWv-Jx_tZQuLIPEY">

What URL did you use?
The URL I posted is only for the library, not the HA Addon.
I have a rudimentary fork in my github for the HA Addon which just changes the origin of the library. But I can not test it at the moment, since my HA device is too far away from my bluetty AC200L, so I have no idea if it will work. If you want to give it a try.
The repo URL for the addon is: [https://github.com/ftrueck/bluetti2mqtt](https://github.com/ftrueck/bluetti2mqtt). There is no guarantee that it will work because i have no idea of docker currently. :-)

ftrueck, 2024-05-07:
> > @tripzero there are controls for:
> > 
> > * AC Output
> > * DC Output
> > * Power Off
> > * Display Timeout
> > 
> > The input can not be changed at the moment.
> 
> Is that an bluetooth API thing or just we lack the right commands? I can't control it from the app, so I'm guessing it's the former...

I think probably both. The protocol is not disclosed from bluetti. So everything needs to be reverse engineered by hand. I do not know if someone already found the needed commands. I guess they exist, since you can control the inputs over bluetooth with the official bluetti app.