warhammerkid/bluetti_mqtt #107: I just installed with misquitto mqtt on home assistant and get the following error when I try to start blueti-mqtt

ceredoandy, 2024-03-14:
We renamed asyncio-mqtt to aiomqtt and released a version 1.0.0 in the process. This is the last release under the asyncio-mqtt name. 
2024-03-14 13:50:27 INFO     Connecting to MQTT broker...
Exception ignored in: <function Client.__del__ at 0x0000023170937100>
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\paho\mqtt\client.py", line 874, in __del__
    self._reset_sockets()
  File "C:\Program Files\Python312\Lib\site-packages\paho\mqtt\client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "C:\Program Files\Python312\Lib\site-packages\paho\mqtt\client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
2024-03-14 13:50:27 INFO     Starting to poll clients...
2024-03-14 13:50:27 ERROR    Crashing with uncaught exception:
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\bluetti_mqtt\mqtt_client.py", line 513, in run
    async with Client(
               ^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\asyncio_mqtt\client.py", line 298, in __init__
    self._client: mqtt.Client = mqtt.Client(
                                ^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'
2024-03-14 13:50:27 INFO     Connecting to clients: ['70:B8:F6:81:A7:66']
2024-03-14 13:50:27 INFO     Shutting down...

shbatm, 2024-03-16:
Seeing the same issue connecting to Home Assistant Addon

chaoszone, 2024-04-05:
Has anyone found a fix yet?

shbatm, 2024-04-05:
Working with my Pull Request here if you want to use that version for now:

https://github.com/warhammerkid/bluetti_mqtt/pull/108

chaoszone, 2024-04-06:
How do I get your version into my HA instance?

ftrueck, 2024-04-09:
I am currently preparing of creating a separate addon and a separate github fork for a more updated version of the underlying library. Not sure how long it will take until I am done but I have the same Issues as you guys and I am kind of tired waiting for the original maintainers to wake up from their winter sleep.