warhammerkid/bluetti_mqtt #114: Bluetooth connection will not initiate with AC500

Wuastr, 2024-05-23:
Help! It seems there is no Bluetooth connection. AC500 will BT connect to Bluetti App. Tried two BT adaptors on my dedicated Surface Laptop.

Log:

2024-05-23 15:23:09 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_CF_D8_C3_4D_34_FC): ['org.bluez.Device1', {}, ['RSSI']]
2024-05-23 15:23:09 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_70_62_6E_FF_53_47): ['org.bluez.Device1', {}, ['TxPower', 'RSSI']]
2024-05-23 15:23:09 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_E4_65_B8_41_6C_DE): ['org.bluez.Device1', {}, ['RSSI']]
2024-05-23 15:23:09 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/00-banner.sh

-----------------------------------------------------------
 Add-on: Bluetti to MQTT
 MQTT bridge between Bluetti and Home Assistant
-----------------------------------------------------------
 Add-on version: 1.4.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.5.4
 Home Assistant Supervisor: 2024.05.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
cont-init: info: /etc/cont-init.d/00-banner.sh exited 0
cont-init: info: running /etc/cont-init.d/01-log-level.sh
cont-init: info: /etc/cont-init.d/01-log-level.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[15:23:52] INFO: Reading configuration settings...
[15:23:52] INFO: Debug mode is enabled.
[15:23:52] INFO: Starting bluetti-mqtt...
2024-05-23 15:23:53 DEBUG    Using selector: EpollSelector
Scanning....
2024-05-23 15:23:53 DEBUG    initial properties: {'/org/bluez': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.AgentManager1': {}, 'org.bluez.ProfileManager1': {}}, '/org/bluez/hci0': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': 'E8:48:B8:C8:20:00', 'AddressType': 'public', 'Name': 'homeassistant #1', 'Alias': 'homeassistant #1', 'Class': 7078156, 'Powered': True, 'PowerState': 'on', 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000184f-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '00001844-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001845-0000-1000-8000-00805f9b34fb', '0000184d-0000-1000-8000-00805f9b34fb', '00001843-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d054B', 'Roles': ['central', 'peripheral'], 'Manufacturer': 93, 'Version': 10}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.BatteryProviderManager1': {}, 'org.bluez.GattManager1': {}, 'org.bluez.AdvertisementMonitorManager1': {'SupportedMonitorTypes': ['or_patterns'], 'SupportedFeatures': []}, 'org.bluez.Media1': {'SupportedUUIDs': ['0000110a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb']}, 'org.bluez.NetworkServer1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 4, 'SupportedIncludes': ['tx-power', 'appearance', 'local-name'], 'SupportedSecondaryChannels': ['1M', '2M', 'Coded'], 'SupportedFeatures': ['CanSetTxPower', 'HardwareOffload'], 'SupportedCapabilities': {'MinTxPower': -19, 'MaxTxPower': 23, 'MaxAdvLen': 251, 'MaxScnRspLen': 251}}}, '/org/bluez/hci0/dev_73_EE_D9_97_A7_FF': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': '73:EE:D9:97:A7:FF', 'AddressType': 'random', 'Alias': '73-EE-D9-97-A7-FF', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -88, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: bytearray(b'\x16\x08\x00\xa5\xe6f\xfc)\x11C')}, 'ServicesResolved': False, 'AdvertisingFlags': bytearray(b'\x1a')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_60_6B_4B_AB_D6_0E': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': '60:6B:4B:AB:D6:0E', 'AddressType': 'random', 'Alias': '60-6B-4B-AB-D6-0E', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -82, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: bytearray(b'\x10\x06\x01\x1e\xb50 \xc3')}, 'TxPower': 12, 'ServicesResolved': False, 'AdvertisingFlags': bytearray(b'\x1a')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci1': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': 'BC:83:85:DC:BD:91', 'AddressType': 'public', 'Name': 'homeassistant', 'Alias': 'homeassistant', 'Class': 7078156, 'Powered': True, 'PowerState': 'on', 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000184f-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '00001844-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001845-0000-1000-8000-00805f9b34fb', '0000184d-0000-1000-8000-00805f9b34fb', '00001843-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d054B', 'Roles': ['central', 'peripheral'], 'Manufacturer': 72, 'Version': 8}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.BatteryProviderManager1': {}, 'org.bluez.GattManager1': {}, 'org.bluez.AdvertisementMonitorManager1': {'SupportedMonitorTypes': ['or_patterns'], 'SupportedFeatures': []}, 'org.bluez.Media1': {'SupportedUUIDs': ['0000110a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb']}, 'org.bluez.NetworkServer1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 5, 'SupportedIncludes': ['tx-power', 'appearance', 'local-name'], 'SupportedFeatures': [], 'SupportedCapabilities': {'MaxAdvLen': 31, 'MaxScnRspLen': 31}}}, '/org/bluez/hci1/dev_73_EE_D9_97_A7_FF': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': '73:EE:D9:97:A7:FF', 'AddressType': 'random', 'Alias': '73-EE-D9-97-A7-FF', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -94, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci1', 'ManufacturerData': {76: bytearray(b'\x16\x08\x00\xa5\xe6f\xfc)\x11C')}, 'ServicesResolved': False, 'AdvertisingFlags': bytearray(b'\x1a')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci1/dev_60_6B_4B_AB_D6_0E': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': '60:6B:4B:AB:D6:0E', 'AddressType': 'random', 'Alias': '60-6B-4B-AB-D6-0E', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -96, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/h [... cut ...]
2024-05-23 15:23:58 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
Found AC5002308000844715: address B8:D6:1A:BE:5D:2E
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

ftrueck, 2024-07-18:
You can not use Bluetti App AND this code at the same time. The bluetooth module allows only one active connection at a time. Did you try to use it without the app open on your phone? There is also a little cooldown time after you closed the app.

Wuastr, 2024-07-18:
Thanks, but I do understand that there is only one BT connection possible at a time. I only mentioned the app connection as background information. I purchased another BT dongle (Digitus Bluetooth 4.0 adapter) to replace the TP Link UB500 and that seems to have resolved the issue.Greg

Wuastr, 2026-10-05:
Update, 5 October 2026: I now have a local prototype of bluetti_mqtt 0.15.0 delivering repeated AC500 and AC300 readings to Home Assistant via MQTT on a NUC.

For the current AC500 failure, BLE service discovery and notification subscription succeeded. The station then sent an encryption challenge beginning with hex 2a2a0104. The plaintext client did not complete that handshake. A later “Service Discovery has not been performed yet” error followed disconnection; it was not the initial failure in this test. This describes the current reproduction, not a retrospective diagnosis of the 2024 log above.

The local client now performs the handshake and encrypts/decrypts the Modbus payloads. Encryption code was adapted from Patrick762/bluetti-bt-lib (MIT, commit 0e11a80484a1c89a992189df050328387b0d1277), retaining attribution and removing session-key debug logging:
https://github.com/Patrick762/bluetti-bt-lib
Official protocol reference:
https://github.com/bluetti-official/bluetti-bluetooth-lib

A separate AC300 problem also needed a fix: its GATT device name contained three trailing NUL characters, causing build_device() to index a failed regex match and terminate the bridge. Normalizing trailing NUL padding and explicitly rejecting invalid names resolved that crash. Embedded NULs remain invalid.

Validation:
- Real AC500 station totals and its connected B300S pack SOC, voltage and cell readings reached MQTT/Home Assistant.
- Real AC300 totals and both connected B300 pack SOC/voltage readings reached MQTT/Home Assistant simultaneously.
- Local tests passed for the captured challenge, premature-ready prevention, fragmented/coalesced encrypted replies, plaintext AC300 handling and malformed challenge checksum rejection. Name-parser regression tests also passed.
- Competing Bluetti BT clients were disabled. Configured polling interval was 60 seconds.

This is a diagnostic prototype, not a merge-ready generic encryption implementation. It currently enables encryption for one configured AC500 address and pins the working NUC adapter (hci1); upstream work should make these choices configurable or detected. Long-term reliability and recovery after a real station shutdown remain unverified. One existing internal-power field also needs separate signed/register decoding investigation. This connection fix does not resolve the station's BAT 104 shutdown issue.

Working local dependency pins: Bleak 0.21.1, asyncio-mqtt 0.12.1, paho-mqtt 1.6.1, cryptography 41.0.7, cffi 1.16.0 and pyasn1 0.6.1. The prototype was prepared with OpenAI Codex assistance and tested on my physical devices.

Would an upstream contribution separating the padded-name fix from configurable encrypted-client support be useful?

Wuastr, 2026-10-05:
Note. All the actual work to find issue and resolve was done by ChatGPT Work including posting here. I just sat back and watched.

closed 2026-10-05 by Wuastr