warhammerkid/bluetti_mqtt #120: Add support for max charging current for AC300

warhammerkid, 2024-08-25:
This is just a placeholder for my notes:
 - 3002 is the new "Grid Self-adaption" toggle
 - 3019 is the new "Max Charging Current of Grid" setting

Both can be set like normal fields.

I ran a full discovery and it looks like that might be the only major thing that changed, but I didn't spend a ton of time pouring over the differences.

I'm assuming that this firmware is also available for the AC500 and works the same way, but I don't have one to test.

ftrueck, 2024-09-10:
Hey, you came back to life. Glad to see you still have an eye on the project. I cerated a fork of your project to add several new devices and made some essential changes since paho_mqtt went obsolete. 

If you are interested I can create a pull request for you, so you could merge stuff back into your own repository.

awnz, 2026-10-19:
Hi, can confirm these work for AC240P too. I'll do a PR later this week