warhammerkid/bluetti_mqtt #123: AC50B

rfcarv, 2024-11-10:
Please add support for Bluetti AC50B.

hamid-elaosta, 2025-06-11:
I spent some time reverse engineering the AC50B this evening. I had to use @nhurman fork or I couldn't communicate with the AC50B: https://github.com/nhurman/bluetti_mqtt

I added a new device named `AC50B` in all the relevant places and created a new device file for it (below), then wrote some Python tools to watch the data printed to the log in hex format so I could spot changes and patterns more easily while I changes thingso n the device; I got as far as the values shown in the code below.

I'm not sure what the correct names are for some of the values, so I've commented them where it was unclear, and since I'm doing this of an evening, I wasn't able to test DC input / PV, I can do that some other time.

Hopefully this is a start for someone to add support properly.

```python
from typing import List
from ..commands import ReadHoldingRegisters
from .bluetti_device import BluettiDevice
from .struct import DeviceStruct

class AC50B(BluettiDevice):
    def __init__(self, address: str, sn: str):
        self.struct = DeviceStruct()

        self.struct.add_uint_field('total_battery_percent', 102)
        self.struct.add_swap_string_field('device_type', 110, 6)
        self.struct.add_sn_field('serial_number', 116)

        self.struct.add_decimal_field('power_generation', 154, 1)  # Total power generated since last reset (kwh)

        self.struct.add_version_field('bcu_version', 6175)

        self.struct.add_uint_field('dc_output_power', 140)

        # Not sure what this value is, it's probably not ac_input_voltage, it changes slightly every now and then, but the value (~233.3) doesn't seem to correspond directly
        # to either the AC input voltage or the AC output voltage shown in the app, which are ~246.5V and ~229V respectively.
        self.struct.add_decimal_field('ac_input_voltage', 100, 1)

        self.struct.add_uint_field('ac_input_power', 146)
        self.struct.add_uint_field('ac_output_power', 142)

        # Not sure what this value corresponds to, but it goes to 01 when the battery and goes to 00 when it is full
        self.struct.add_bool_field('charging', 103)

        # Both registers display the same value, time in hours, multiplied by 10, 40.9 is 409
        self.struct.add_decimal_field('battery_time_remaining', 104, 1)
        self.struct.add_decimal_field('battery_time_remaining_2', 105, 1)

        # 20: AC cable disconnected
        # 22: AC cable connected
        self.struct.add_uint_field('ac_related_1', 123)

        #
        # =====================================================================================
        #
        # Register 124: Device State Decoder
        #
        # The final decimal value is a combination of two independent bytes, calculated as:
        # Final Value = (Value of First Byte * 256) + (Value of Second Byte)
        #
        # ---------------------------------
        # --- First Byte (for Device Modes)
        # ---------------------------------
        # This is a bitmask of the following flags (values are added together):
        #   - Value 2: DC Eco Mode is ON
        #   - Value 4: Power Lifting is ON
        #   - Value 8: AC Eco Mode is ON
        #
        # ------------------------------------
        # --- Second Byte (for Output Status)
        # ------------------------------------
        # This is a bitmask of a base value plus the active output flags:
        #   - Base Value 1: Indicates the unit is ON
        #   - Flag 2: AC Output is ON
        #   - Flag 4: DC Output is ON
        #
[... 78 lines cut ...]
            return self.polling_commands

```
Output:
```
{
    "total_battery_percent": 99,
    "device_type": "AC50B",
    "serial_number": [removed],
    "power_generation": Decimal("1.4"),
    "dc_output_power": 0,
    "ac_input_voltage": Decimal("233.3"),
    "ac_input_power": 0,
    "ac_output_power": 0,
    "charging": False,
    "battery_time_remaining": Decimal("99.9"),
    "battery_time_remaining_2": Decimal("99.9"),
    "ac_related_1": 32,
    "device_state": 1,
    "charging_mode": 0,
    "ac_related_2": 0
}
```

I haven't looked at the other ranges yet so I'm sure there's lots more to find, also I didn't look at any controls yet.

nhurman, 2025-06-12:
I would encourage you to take a look at the V2Device class in the fork and see if this works for your AC50B. It seems (at least back in January) that all the devices using encryption were standardized on that protocol.

arc2bk, 2025-09-21:
I could integrate the code with the nhurman code and ran a few tests. I had to disable the "charging_mode" or I would get an error due to the enum-type.
I also added the DC_input_power (essentially solar generation) as register 144. The "power_generation" is actually the cummulated energy, as already noted in the comment.

The code now looks like this (essentially one line for DC_input_power added and one line commented. I could test for a few days now and it seems to run smoothly.

`from typing import List
from ..commands import ReadHoldingRegisters
from .bluetti_device import BluettiDevice
from .struct import DeviceStruct

class AC50B(BluettiDevice):
    def __init__(self, address: str, sn: str):
        self.struct = DeviceStruct()

        self.struct.add_uint_field('total_battery_percent', 102)
        print(self.struct)

        self.struct.add_swap_string_field('device_type', 110, 6)
        self.struct.add_sn_field('serial_number', 116)

        self.struct.add_decimal_field('power_generation', 154, 1)  # Total power generated since last reset (kwh)

        self.struct.add_version_field('bcu_version', 6175)

        self.struct.add_uint_field('dc_input_power', 144)	#added
        self.struct.add_uint_field('dc_output_power', 140)

        # Not sure what this value is, it's probably not ac_input_voltage, it changes slightly every now and then, but the value (~233.3) doesn't seem to correspond directly
        # to either the AC input voltage or the AC output voltage shown in the app, which are ~246.5V and ~229V respectively.
        self.struct.add_decimal_field('ac_input_voltage', 100, 1)

        self.struct.add_uint_field('ac_input_power', 146)
        self.struct.add_uint_field('ac_output_power', 142)

        # Not sure what this value corresponds to, but it goes to 01 when the battery and goes to 00 when it is full
        self.struct.add_bool_field('charging', 103)

        # Both registers display the same value, time in hours, multiplied by 10, 40.9 is 409
        self.struct.add_decimal_field('battery_time_remaining', 104, 1)
        self.struct.add_decimal_field('battery_time_remaining_2', 105, 1)

        # 20: AC cable disconnected
        # 22: AC cable connected
        self.struct.add_uint_field('ac_related_1', 123)

        #
        # =====================================================================================
        #
        # Register 124: Device State Decoder
        #
        # The final decimal value is a combination of two independent bytes, calculated as:
        # Final Value = (Value of First Byte * 256) + (Value of Second Byte)
        #
        # ---------------------------------
        # --- First Byte (for Device Modes)
        # ---------------------------------
        # This is a bitmask of the following flags (values are added together):
        #   - Value 2: DC Eco Mode is ON
        #   - Value 4: Power Lifting is ON
        #   - Value 8: AC Eco Mode is ON
        #
        # ------------------------------------
        # --- Second Byte (for Output Status)
        # ------------------------------------
        # This is a bitmask of a base value plus the active output flags:
        #   - Base Value 1: Indicates the unit is ON
        #   - Flag 2: AC Output is ON
        #   - Flag 4: DC Output is ON
        #
        #   Examples:
[... 55 lines cut ...]
        # 2: Turbo
        #self.struct.add_uint_field('charging_mode', 160)
        #problem mit enum

        # 0: AC cable disconnected
        # 3: AC cable connected
        self.struct.add_uint_field('ac_related_2', 161)

        super().__init__(address, 'AC50B', sn)

    @property
    def polling_commands(self) -> List[ReadHoldingRegisters]:
            return [
                ReadHoldingRegisters(100, 62),
            ]
            

    @property
    def logging_commands(self) -> List[ReadHoldingRegisters]:
        
            #ReadHoldingRegisters(100, 62),
            #ReadHoldingRegisters(6100, 100),
            
            return self.polling_commands
`