warhammerkid/bluetti_mqtt #127: new firmware with encryption EB3A

brunokelo, 2025-01-12:
Does anyone have any tips/suggestions on how to recover information from the eb3a models that are coming with the new firmware, it seems that there is some type of encryption in the bluetouch connection that makes it impossible to read the information sent by the box.

error log:

  **Waiting for connection to ED:25:22:E7:4F:8D to start polling...
2025-01-12 15:44:58 DEBUG    received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_68_F9_24_60_52_0F', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 68:F9:24:60:52:0F)>, 'AddressType': <dbus_fast.signature.Variant ('s', random)>, 'Alias': <dbus_fast.signature.Variant ('s', 68-F9-24-60-52-0F)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -85)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {76: <dbus_fast.signature.Variant ('ay', bytearray(b'\x10\x05\x02\x18\xfaG\xe6'))>})>, 'TxPower': <dbus_fast.signature.Variant ('n', 24)>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>, 'AdvertisingFlags': <dbus_fast.signature.Variant ('ay', bytearray(b'\x1a'))>}, 'org.freedesktop.DBus.Properties': {}}]
2025-01-12 15:44:59 DEBUG    Waiting for connection to ED:25:22:E7:4F:8D to start polling...
2025-01-12 15:45:00 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_C3_F4_AD_10_79): ['org.bluez.Device1', {'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {117: <dbus_fast.signature.Variant ('ay', bytearray(b'B\x04\x01 f\x19\x0f\x00\x00\x012\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>})>}, []]
2025-01-12 15:45:00 DEBUG    Waiting for connection to ED:25:22:E7:4F:8D to start polling...
2025-01-12 15:45:00 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_C3_F4_AD_10_79): ['org.bluez.Device1', {'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {117: <dbus_fast.signature.Variant ('ay', bytearray(b'B\x04\x01\x01f\x00\xc3\xf4\xad\x10y\x02\xc3\xf4\xad\x10x\x01O\xfb\x00\x00\x00\x00'))>})>}, []]
2025-01-12 15:45:01 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_C3_F4_AD_10_79): ['org.bluez.Device1', {'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {117: <dbus_fast.signature.Variant ('ay', bytearray(b'B\x04\x01 f\x19\x0f\x00\x00\x012\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>})>}, []]
2025-01-12 15:45:01 DEBUG    Waiting for connection to ED:25:22:E7:4F:8D to start polling...
2025-01-12 15:45:01 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_22_7D_79_34_D7): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -69)>}, []]
2025-01-12 15:45:02 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_C3_F4_AD_10_79): ['org.bluez.Device1', {'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {117: <dbus_fast.signature.Variant ('ay', bytearray(b'B\x04\x01\x01f\x00\xc3\xf4\xad\x10y\x02\xc3\xf4\xad\x10x\x01O\xfb\x00\x00\x00\x00'))>})>}, []]
2025-01-12 15:45:02 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_22_7D_79_34_D7): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -61)>}, []]
2025-01-12 15:45:02 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_22_7D_79_34_D7): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -69)>}, []]
2025-01-12 15:45:02 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_22_7D_79_34_D7): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -61)>}, []]
2025-01-12 15:45:02 DEBUG    Connecting to device @ ED:25:22:E7:4F:8D
2025-01-12 15:45:02 DEBUG    Connecting to BlueZ path /org/bluez/hci0/dev_ED_25_22_E7_4F_8D
2025-01-12 15:45:02 DEBUG    Waiting for connection to ED:25:22:E7:4F:8D to start polling...
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', True)>}, []]
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_C3_F4_AD_10_79): ['org.bluez.Device1', {'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {117: <dbus_fast.signature.Variant ('ay', bytearray(b'B\x04\x01 f\x19\x0f\x00\x00\x012\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>})>}, []]
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D): ['org.bluez.Device1', {'ServicesResolved': <dbus_fast.signature.Variant ('b', True)>}, []]
2025-01-12 15:45:03 INFO     Connected to device: ED:25:22:E7:4F:8D
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D/service001d/char001e): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_fast.signature.Variant ('b', True)>}, []]
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D/service001d/char001e): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'**\x01\x04\xf2\x8c\xe9\x8a\x02\xf6'))>}, []]
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_22_7D_79_34_D7): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -69)>}, []]
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_C3_F4_AD_10_79): ['org.bluez.Device1', {'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {117: <dbus_fast.signature.Variant ('ay', bytearray(b'B\x04\x01\x01f\x00\xc3\xf4\xad\x10y\x02\xc3\xf4\xad\x10x\x01O\xfb\x00\x00\x00\x00'))>})>}, []]
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D/service001d/char001e): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'**\x01\x04\xf2\x8c\xe9\x8a\x02\xf6'))>}, []]
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D): ['org.bluez.Device1', {'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>}, []]
2025-01-12 15:45:03 DEBUG    Device disconnected (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D)
2025-01-12 15:45:03 DEBUG    _cleanup_all(/org/bluez/hci0/dev_ED_25_22_E7_4F_8D)
2025-01-12 15:45:03 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D/service001d/char001e): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_fast.signature.Variant ('b', False)>}, []]
2025-01-12 15:45:03 DEBUG    Disconnecting (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D)
2025-01-12 15:45:03 DEBUG    already disconnected (/org/bluez/hci0/dev_ED_25_22_E7_4F_8D)
/usr/local/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/client.py:165: DeprecationWarning: The 'warn' function is deprecated, use 'warning' instead
  logging.warn(f'Delayed reconnect to {self.address} after error')
2025-01-12 15:45:03 WARNING  Delayed reconnect to ED:25:22:E7:4F:8D after error
2025-01-12 15:45:03 DEBUG    Needed to disconnect due to error: Not connected
2025-01-12 15:45:04 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_22_7D_79_34_D7): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -61)>}, []]
[... cut ...] 4\xad\x10x\x01O\xfb\x00\x00\x00\x00'))>})>}, []]
2025-01-12 15:45:06 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_22_7D_79_34_D7): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -69)>}, []]
2025-01-12 15:45:06 DEBUG    received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_45_20_68_EE_B7_27): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -66)>}, []]**

thanks

bres55, 2025-05-14:
I've tried installing Nhurman version. I get errors running it

C:\Windows\System32>bluetti-mqtt --scan
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\Python313\Scripts\bluetti-mqtt.exe\__main__.py", line 4, in <module>
    from bluetti_mqtt.server_cli import main
  File "C:\Program Files\Python313\Lib\site-packages\bluetti_mqtt\server_cli.py", line 9, in <module>
    from bluetti_mqtt.bluetooth import scan_devices
  File "C:\Program Files\Python313\Lib\site-packages\bluetti_mqtt\bluetooth\__init__.py", line 7, in <module>
    from .client import BluetoothClient
  File "C:\Program Files\Python313\Lib\site-packages\bluetti_mqtt\bluetooth\client.py", line 9, in <module>
    from .encryption import Connection, PassthroughConnection, EncryptedConnection
  File "C:\Program Files\Python313\Lib\site-packages\bluetti_mqtt\bluetooth\encryption.py", line 60, in <module>
    from cryptography.exceptions import InvalidSignature
ModuleNotFoundError: No module named 'cryptography'

msmialow, 2025-07-12:
Note, there was an issue with nhurman/bluetti_mqtt@bf037c3.

I ended up having to go in and modify line 161 in server_cli.py from `def main(argv):` to `
def main(argv=None):` in order to get the final bluetti-mqtt binary to run.  Now it works for me perfectly and I can do what I please with this device in home assistant.  My EB3A unit is brand new out of box as of 2 days ago AND I updated everything firmware via the app before finding out this was a thing, so it should work for most people once that line is changed (unless there are some unintended consequences).

Note, I don't want to do a PR or anything as I don't really know what I'm doing here, I know just enough to debug and web search until something works for me. @nhurman, I'm assuming you'll know if my change was correct or wildly stupid, so if the later, please let us know and thanks for getting the encryption working.

@bres55, my understanding is that installing from git repos does not use the reference.txt file to install the dependencies.  That's likely your issue.  You need to install them manually with `python -m pip install -r requirements.txt`.  At least I think the same command would work.  No clue why you are trying something like this on windows, but, good luck with that I guess.  Linux is in fact a thing you know. For those installing in linux you should make sure you are using a virtual environment properly and not installing over any system packages as these are version locked in the text file.  Ask me how I know.  I had an especially annoying time getting pip to run in the venv and not system wide on a headless raspbian install on a pi5.  In the end, I had to install pip in the venv manually and then only reference it in the venv with `python -m pip` and never directly.  A bit of a PITA, but now working in the end.