warhammerkid/bluetti_mqtt #129: What would you need from Bluetti to make it work flawless?

SelfmadestromDE, 2025-03-11:
Hey guys,

im in contact with bluetti to add a "official" possibility to use the units via MQTT. My question to the maintainers:

What you guys would need from Bluetti to actually support every unit with all its functionalitys? They mentioned, that writing its own application would be to time consuming, maybe we can go this way more easier.

Thanks
Erik

MegaBarf, 2025-03-14:
Hi

Please also ask at: https://github.com/SSMCD/bluetti2mqtt (which I'm still using for a EP500Pro)
and: https://github.com/Patrick762/hassio-bluetti-bt (which I plan to use for my new EP2000 System)

MegaBarf, 2025-03-14:
Major thing that would be needed would be the Modbus registers and their definitions.

Or a way to directly input a local mqtt server into the units while being connected to WLAN (especially without internet access in that network!!)

For myself with the EP2000 a RS485 Modbus interface would be best. I don't want to have unnecessary wireless connections.

SelfmadestromDE, 2025-03-14:
Thanks @MegaBarf

They dont will develope any new things from what i hear now. We depend on the community add ons. But i just want them to make clear, what thinks this project need to be good maintainable.

I will have a look, if they can publish the Modbus registers and their definitions.

SelfmadestromDE, 2025-04-08:
@MegaBarf @alexnathanson @warhammerkid @jretz 

Would it be possible to write a guide on how exactly you guys add support for new units? Or is there already any.

Maybe than it would be more easier to actually get what we want from the community itself. Many People out there have the units, but dont know how to add this to the tools, myself included.

tripzero, 2025-04-15:
What I really need for my automation to work is the ability change the "Custom UPS" SOC Setting.  Here's my use-case:

Solar charges the battery during the day and during off peak, the grid charges to SOC_Setting_High.  On cloudy days, I want SOC_Setting_high to be 90-100% because there will not be much solar input.  On sunny days, I want to set SOC_Setting_High to 40-60 depending on the month of the year (angle of the sun, total hours of sunlight, etc).  I can use machine learning to predict how much solar power I will generate every hour 24 hours in advance (using weather data) with pretty good accuracy and determine how high the SOC_Setting_High should be.  The only problem, is I can't actually set it programatically. 

So basically we need to be able to read every information available in the app and write every setting available in the app.  Whether that's available through MQTT or via a web API doesn't matter.  We just need something.

SelfmadestromDE, 2025-04-16:
@tripzero Actually this "Issue" was open to bring this community project on a great track. But since you mentioned it, bluetti actually started a survey for they official API.

Please share your Feedback here:

https://docs.google.com/forms/d/e/1FAIpQLSfM7QH9PHkGEMApVMHpHLxqmNKNxsDIUcda3l8XeqF_hP3SYg/viewform

bres55, 2025-05-11:
I just want to be able to see within Home Assistant, my AC180, and EB3A, SOC, and be able to turn on/off, AC and DC output. Maybe, Panel Data. I am so disappointed my replacement AC180 will not integrate, like the old one used to.

chpego, 2026-09-29:
@MegaBarf - two different things, and I will be straight about which is which.

Your **EP500P** is the one that works today: that model serves Modbus TCP on its own IoT module over Wi-Fi, no cloud and no Bluetooth, and the register map is confirmed on two real units at [bluetti-community/bluetti-registers](https://github.com/bluetti-community/bluetti-registers/issues/35). Enable it on the unit's local web page (Settings -> Modbus TCP, port 502) and it reads straight away.

Your **EP2000** is the open question, and the honest state is that nobody has got Modbus TCP out of one yet. BLUETTI's own register list covers the model, but the two units anyone has looked at had no Modbus TCP entry on their web page at all - one owner with firmware V503222/V503021 had completed every available update and still only had Bluetooth, WiFi and WifiAP in Settings. So the firmware that serves it may simply not have shipped for this model.

That is exactly why a third unit would be useful. Your web page's Settings list, and the firmware versions it shows for the EP2000 and the EBOX, either match what we have or tell us a newer build exists: [bluetti-registers#42](https://github.com/bluetti-community/bluetti-registers/issues/42). There is also an RS485 port on the EMS that an owner is preparing to listen to, but that track has not produced anything yet either.