warhammerkid/bluetti_mqtt #136: Crash on HA OS 16.2 with AC200M: TypeError: 'NoneType' object is not subscriptable

TeleAndy, 2025-09-09:
**Describe the bug**
After updating to Home Assistant OS 16.2 (aarch64 / HA Green), the Bluetti2MQTT add-on no longer works with my Bluetti AC200M.  
The add-on starts, connects to MQTT and to the device, but then crashes with an exception.

**Environment**
- Add-on version: 1.4.1
- Home Assistant OS: 16.2 (aarch64 / HA Green)
- Home Assistant Core: 2025.8.3
- Supervisor: 2025.09.0
- Device: Bluetti AC200M
- Connection: Bluetooth
- MQTT broker: core-mosquitto

**Logs**

2025-09-09 09:23:58 INFO     Connecting to MQTT broker…
2025-09-09 09:23:58 INFO     Starting to poll clients…
2025-09-09 09:23:58 INFO     Connecting to clients: [‘04:7F:0E:A4:0D:BC’]
2025-09-09 09:23:59 INFO     Connected to MQTT broker
2025-09-09 09:24:05 INFO     Connected to device: 04:7F:0E:A4:0D:BC
2025-09-09 09:24:05 INFO     Device 04:7F:0E:A4:0D:BC has name: AC200M2022002008344
2025-09-09 09:24:05 ERROR    Crashing with uncaught exception:
Traceback (most recent call last):
File “/usr/local/lib/python3.10/site-packages/bluetti_mqtt/device_handler.py”, line 31, in run
await asyncio.gather(*(polling_tasks + pack_polling_tasks + [manager_task]))
File “/usr/local/lib/python3.10/site-packages/bluetti_mqtt/device_handler.py”, line 45, in _poll
device = self._get_device(address)
File “/usr/local/lib/python3.10/site-packages/bluetti_mqtt/device_handler.py”, line 103, in _get_device
self.devices[address] = build_device(address, name)
File “/usr/local/lib/python3.10/site-packages/bluetti_mqtt/bluetooth/init.py”, line 28, in build_device
if match[1] == ‘AC200M’:
TypeError: ‘NoneType’ object is not subscriptable

**Expected behavior**
The add-on should recognize the AC200M model and provide MQTT sensors, as it did before the update (HA OS 6.1 worked fine).

**Additional context**
- The device connects successfully via Bluetooth and the name is detected (`AC200M2022002008344`), but the `build_device` function seems to fail with `NoneType`.
- It looks like model detection fails for AC200M on newer HA OS versions.

**Possible solution**
Maybe add an explicit model override option in the config:
```yaml
devices:
  - address: "04:7F:0E:A4:0D:BC"
    name: "AC200M2022002008344"
    model: "AC200M"

alterHase77, 2025-09-13:
I added

devices:
  - address: "04:7F:0E:A4:0D:BC"
    name: "AC3002226000036553"
    model: "AC300"

to the HA config.yaml but ist still doesn't work

F0551L, 2025-09-18:
@alterHase77 and @TeleAndy  - see my reply on #135 , if you haven't previously trusted the connection with bluetoothctl over SSH or directly in the HA CLI, doing that may solve this for you (I'm using bluetti-bit not this addon but it seems to be a hardware-level resolution)